verilog-to-routing/vtr-verilog-to-routing #2403: Warnings should be fixed in vqm2blif

vaughnbetz, 2023-09-28:
We should be warning clean, but vqm2blif isn't.  See https://github.com/verilog-to-routing/vtr-verilog-to-routing/actions/runs/6333593038/job/17201935175 for a CI run that shows the warnings.

#### Expected Behaviour
No compiler warnings.

#### Current Behaviour
vqm2blif compiler warnings

02:02:20 | [ 13%] [BISON][VqmParser] Building parser with bison 3.8.2
02:02:20 | /root/vtr-verilog-to-routing/vtr-verilog-to-routing/libs/libvqm/vqm_parser.y:45.1-14: warning: deprecated directive: '%error-verbose', use '%define parse.error verbose' [-Wdeprecated]
02:02:20 |    45 | %error-verbose
02:02:20 |       | ^~~~~~~~~~~~~~
02:02:20 |       | %define parse.error verbose
02:02:20 | [ 13%] [FLEX][VqmLexer] Building scanner with flex 2.6.4
02:02:20 | [ 13%] Building CXX object libs/EXTERNAL/capnproto/c++/src/kj/CMakeFiles/kj-async.dir/async-win32.c++.o
02:02:20 | [ 13%] Building CXX object libs/EXTERNAL/capnproto/c++/src/kj/CMakeFiles/kj-async.dir/async-io-win32.c++.o
02:02:20 | [ 13%] Building C object abc/CMakeFiles/libabc.dir/src/base/abci/abcDress2.c.o
02:02:20 | [ 13%] Building C object abc/CMakeFiles/libabc.dir/src/base/abci/abcDress3.c.o
02:02:20 | /root/vtr-verilog-to-routing/vtr-verilog-to-routing/libs/libvqm/vqm_parser.y: warning: fix-its can be applied.  Rerun with option '--update'. [-Wother]
02:02:20 | [ 13%] Building CXX object libs/EXTERNAL/libcatch2/src/CMakeFiles/Catch2.dir/catch2/internal/catch_random_number_generator.cpp.o
02:02:20 | [ 13%] Building CXX object libs/libvqm/CMakeFiles/libvqm.dir/vqm_common.c.o
02:02:20 | [ 13%] Building C object abc/CMakeFiles/libabc.dir/src/base/abci/abcDsd.c.o
02:02:20 | [ 13%] Building C object abc/CMakeFiles/libabc.dir/src/base/abci/abcEco.c.o
02:02:20 | [ 13%] Building C object abc/CMakeFiles/libabc.dir/src/base/abci/abcExact.c.o
02:02:21 | [ 13%] Building CXX object libs/libvqm/CMakeFiles/libvqm.dir/vqm_dll.cpp.o
02:02:21 | [ 13%] Building CXX object libs/libvqm/CMakeFiles/libvqm.dir/vqm_lexer.gen.c.o
02:02:21 | /root/vtr-verilog-to-routing/vtr-verilog-to-routing/libs/libvqm/vqm_dll.cpp: In function 'int vqm_get_error_message(char*, int)':
02:02:21 | /root/vtr-verilog-to-routing/vtr-verilog-to-routing/libs/libvqm/vqm_dll.cpp:120:40: warning: passing argument 1 to restrict-qualified parameter aliases with argument 2 [-Wrestrict]
02:02:21 |   120 |   strcpy(message_buffer, message_buffer);
02:02:21 |       |                          ~~~~~~~~~~~~~~^
02:02:21 | [ 13%] Building CXX object libs/libvqm/CMakeFiles/libvqm.dir/vqm_parser.gen.c.o

vaughnbetz, 2023-10-05:
@kimiatkh @Srivat97 : I am not sure who is the best person to fix these warnings (not sure if they are in the new hard block code or S10 code or something else) but it would be great if one of you could fix them.

kimiatkh, 2023-10-05:
@vaughnbetz I can take care of that. Should be easy to fix. I think some of the warnings may be due to a recent compiler version upgrade, if there has been one.

AlexandreSinger, 2024-05-16:
@vaughnbetz I think this issue can be closed since it was resolved in Navid's most recent PR. PR #2549