verilog-to-routing/vtr-verilog-to-routing #2412: Fix warning in FASM

vaughnbetz, 2023-10-06:
Recent CI shows this warning in fasm:

/home/runner/work/vtr-verilog-to-routing/vtr-verilog-to-routing/utils/fasm/src/fasm.cpp:572:24: warning: loop variable ‘param’ creates a copy from type ‘const std::__cxx11::basic_string<char>’ [-Wrange-loop-construct]
  572 |         for(const auto param : vtr::split(fasm_params_str, "\n")) {
      |                        ^~~~~
/home/runner/work/vtr-verilog-to-routing/vtr-verilog-to-routing/utils/fasm/src/fasm.cpp:572:24: note: use reference type to prevent copying
  572 |         for(const auto param : vtr::split(fasm_params_str, "\n")) {
      |                        ^~~~~
      |                        &

@duck2 : can you fix this?  I think it just needs a ref (&).

hzeller, 2023-10-06:
In general, not only using string references, but also using more `std::string_view` would help avoid a lot of allocating new strings. I see that there are a lot of functions that take a `std::string` as value parameter and return `std::vector<std::string>` (e.g. [`split_fasm_entry()`](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/15c5340681d14acd4aae2fc367191d66114f9a2b/utils/fasm/src/fasm_utils.h#L22-L24)).

(And for fast fasm parsing, I'd like to plug this thing I did a while back https://github.com/hzeller/simple-fasm - feel free to just copy into the project).

vaughnbetz, 2023-10-06:
@hzeller: if that new fasm works better I'm happy to see it integrated, but I think you'd have to drive it since I don't have any fasm experts on my team.

AlexandreSinger, 2024-04-16:
This warning has been resolved in PR #2535 

@vaughnbetz should we close this issue?

closed 2024-04-16 by vaughnbetz