verilog-to-routing/vtr-verilog-to-routing #2413: Flat router delays in timing report are inaccurate due to treating clock as global net

saaramahmoudi, 2023-10-06:
Treating clock as global nets will make net_delay zero. Using two-stage router, the delay of an interconnect is calculated by adding 'driver delay', 'cluster net delay' and 'sink delay', which driver and sink delays are intra-cluster delay between source pin to cluster pin and cluster pin to sink pin. However, flat router uses atom net and it is from one primitive to another, which means we don't need to add 'driver delay' and 'sink delay' anymore and we simply use net_delay for it. 

#### Current Behaviour
The following attach screenshot shows the difference between timing reports in two-stage (left side) router and flat router (right side). 
[timing_report_screenshot](https://github.com/verilog-to-routing/vtr-verilog-to-routing/assets/29703355/4b845ca8-2155-4c43-822c-bbfccba1a14a)
We can see in two-stage router we have intra 'io' routing is calculated (0.042) correctly based on architecture file and the same delay in flat router reported as 0 since net_delay is 0 as it is a clock. 

#### Expected Behaviour
intra 'io' routing should be considered in flat router scenario, as well. 

#### Steps to Reproduce
1. Clone and build master branch
2. Architecture file: vtr_flow/arch/timing/k6_frac_N10_40nm.xml, Design file: vtr_flow/benchmarks/blif/clma.blif

#### Context
@tpagarani @amin1377 @vaughnbetz

vaughnbetz, 2025-05-26:
I'm not sure this is a real problem. If we don't route the clocks our delays will be inaccurate so I am not sure adding in a little bit of the delay in this case is worth it.  @amin1377 : any thoughts?

amin1377, 2025-05-26:
My only concern is that the results are inconsistent with the two-stage router. This delay is included in the two-stage router but treated as zero when the flat router is used. That said, I agree that the delay is small and doesn't make any real difference.