verilog-to-routing/vtr-verilog-to-routing #2416: Some koios circuits failing with parmys

duck2, 2023-10-10:
`proxy_2.v` and `proxy_3.v` do not seem to go through the synthesis step when running `vtr_reg_weekly/koios`. Here is the output file for proxy_2: [parmys.out.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/12860171/parmys.out.txt)

Looks like the file didn't capture the entire output, but yosys seems to complain about `ERROR: Multiple edge sensitive events found for this signal!`

alirezazd, 2023-10-16:
@poname Please take a look at this.

poname, 2023-10-16:
@alirezazd @duck2 I believe this is not related to `Parmys` itself and `Yosys` is throwing the exception---before `Parmys`. So, whether `Yosys` has an issue or the input circuits should be implemented differently.

A few external discussions can be found here:
https://www.reddit.com/r/yosys/comments/3083i8/multiple_edge_sensitivity_error/
https://stackoverflow.com/questions/72483663/yosys-multiple-edge-sensitivities-for-asynchronous-reset
https://electronics.stackexchange.com/questions/517968/error-handling-two-posedge-signals-in-always-block

closed 2025-06-10 by github-actions[bot]