verilog-to-routing/vtr-verilog-to-routing #2424: Regarding the issue of increasing voltage leading to a decrease in power consumption

luck-codeer, 2023-10-23:
I conducted two measurements of the same FPGA architecture at low temperatures with power supply voltages of 0.7V and 0.8V, the value of vth0 was also changed to 0.3 in the 22pm process, and the corresponding 22nm.xml file was generated. and found that the power consumption was actually lower at 0.8V than at 0.7V. I don't quite understand if there is an issue with my measurement.Below are the results of my measurements:
[image]
[image]

luck-codeer, 2023-10-31:
When running generate_cmos_tech_data.pl with hspice 2017, you may come across an issue where the current values need to be converted into exponential notation with the "e" notation. Additionally, certain lines may contain values in exponential notation that need to be preserved. If the power consumption results at the bottom of the output are missing, it may require modifying a Python file as some code may only be compatible with Python 2 versions.
[image]

vaughnbetz, 2023-10-31:
Thanks @luck-codeer . Do you have a code patch?  If so I suggest making a PR with that code so we can patch the script. At the minimum, adding a comment to the script would be helpful, but if the underlying vulnerability can be fixed that is best of all.  You just need to confirm the patched script generates correct results (or at least as correct as the prior script).

luck-codeer, 2023-11-02:
Sure. Professor Betz, I have submitted a PR regarding this issue. you could take a look and see if it's helpful.

luck-codeer, 2023-11-02:
Professor Betz，I have a few questions to consult with you. I would appreciate it if you could provide me with some guidance when you have some free time.
When I conducted experiments at low temperatures and measured the leakage current of transistors using HSPICE, I found that the leakage current may not necessarily decrease at low temperatures. In fact, it can be even higher than at room temperature when the threshold voltage (vth0) is very small, such as 0.2. The testing procedure and results are shown in the following figure.
This is the SPICE program used to measure the leakage current.
[image]
This is the leakage power measured at room temperature.
[image]
This is the leakage power measured at low temperature (vth0=0.2).
[image]
I came across a formula in a paper(https://ieeexplore.ieee.org/abstract/document/1250885) that suggests this possibility, 
[image]
which seems to contradict the description I read before about low temperatures reducing static power consumption. I think the elimination of static power consumption at low temperatures may require the threshold voltage not to be too small. However, I'm not entirely sure, so I would like to ask for your advice. 
Additionally, I ran some benchmarks using a specific FPGA architecture with the command '../scripts/run_vtr_task.py power/power_nofrac_22nm -j 40' and this is This is the parse_results.txt obtained. 
[image]
and I observed that in the power estimation, static power dominates. for example, consider the LU8PEEng circuit shown below.
[image]
I'm wondering if in most cases, the power consumption of an FPGA running a corresponding circuit comes mainly from static power. Could you please provide me with some advice?

vaughnbetz, 2023-11-05:
How much power consumption is static and how much is dynamic varies; it depends on T and the activity and clock frequency and size of the design.  For most FPGA designs running at full speed in a mostly full chip, dynamic power is larger than static power (typical: 2/3 dynamic) but there is significant variation.  Make sure your design fills most of the chip resources, is running at or near its maximum frequency and has reasonable activities (a typical toggle rate is in the 12% - 50% range).

I'm not sure why the SPICE decks are showing decreased leakage with temperature. At least in advanced processes, subthreshold leakage is the biggest contributor to static power, and it increases with temperature. At very low temperatures (e.g. I think you looked at -200) I am not sure of exactly what happens to gate leakage etc., but certainly from -40 C to +100 C static power will increase with temperature.

When you give Vth0 values are you saying you are changing the threshold voltage between your two simulations?  That would of course greatly change the static power.