verilog-to-routing/vtr-verilog-to-routing #2428: Clang CI tests are failing

saaramahmoudi, 2023-11-02:
CI tests related to Clangs (Clang 11, Clang 12, Clang 13, Clang 14) are failing for all our pull requests now. The error for one of them is shown in the picture below. 

[image]

#### Expected Behaviour

CI tests related to Clang should be green and passed. As an example, you can look into [this PR](https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/2348).

vaughnbetz, 2023-11-04:
@duck2 can you take a look?  Maybe an update to catch2 brought in some construct that Clang doesn't understand without a command line argument?  Not sure what else could cause this ...

jgoeders, 2023-11-14:
Fixed with #2438 and #2437

closed 2023-11-14 by jgoeders