verilog-to-routing/vtr-verilog-to-routing #2436: catch2 not including <stdint.h>, causing compiler errors with Clang

vaughnbetz, 2023-11-09:
Recently Clang has failed to compile the catch2 library.  This appears to be a valid error due to catch2 not including <stdint.h>

#### Expected Behaviour
VTR should compile under Clang.

#### Current Behaviour
```
[ 65%] Building CXX object libs/EXTERNAL/libcatch2/src/CMakeFiles/Catch2.dir/catch2/catch_test_case_info.cpp.o
In file included from /home/runner/work/vtr-verilog-to-routing/vtr-verilog-to-routing/libs/EXTERNAL/libcatch2/src/catch2/catch_test_case_info.cpp:8:
/home/runner/work/vtr-verilog-to-routing/vtr-verilog-to-routing/libs/EXTERNAL/libcatch2/src/catch2/../catch2/catch_test_case_info.hpp:39:37: error: unknown type name 'uint8_t'
    enum class TestCaseProperties : uint8_t {
                                    ^
```

#### Possible Solution
We could add the missing include. I am not sure how often we pull down new versions of catch2; if often, we should probably try to fix the source repo.

#### Steps to Reproduce
Run CI.

#### Context
All our CI tests have failures.

vaughnbetz, 2023-11-09:
Adding @duck2 as the likely driver of this.

jgoeders, 2023-11-10:
@duck2 It seems the newer version is breaking a few things: https://github.com/verilog-to-routing/vtr-verilog-to-routing/actions/runs/6829459730/job/18575644801?pr=2437#step:5:2005.  

I'm going to try and fix them on my PR today.

PR #2437 (2023-11-13): Update catch2 and move to submodule. Add dependabot for submodules

#### Description
Replace catch2 with a submodule, which is an updated version of catch2.  This PR also contains some fixes for the newer catch2 version.

This is the first time submodules are being introduced into the VTR code base; however, when linking against an externally developed library like catch2, they are more appropriate than subtrees (subtrees bring in all files and history into the repo, which isn't what we want here).

One issue I encountered is we have one subtree project (parmys) that has submodules within it.  [... cut]
closed 2023-11-13 by vaughnbetz