verilog-to-routing/vtr-verilog-to-routing #2439: Update documentation for .place file to say the block name can be a cluster name or a primitive name within it

vaughnbetz, 2023-11-13:
#### Expected Behaviour
A .place file can (I think but good to confirm) use either a clustered block name or the name of a primitive within it. A primitive is often more convenient since its name doesn't change.

#### Current Behaviour
Our .place file documentation appears to be out of date. It says we need to use the cluster name for a block.

#### Possible Solution
Update the .place file documentation.

ZohairZaidi, 2024-08-18:
is this update to the .place doc appropriate? I will immediately submit a pull request upon your approval.
- **The `block_name` can be either:**
  - **The name of a clustered block, as given in the input `.net` formatted netlist.**
  - **The name of a primitive within a clustered block**

- `x` and `y` are the row and column in which the block is placed, respectively.

KA7E, 2024-08-19:
VPR tracks cluster names in two places: the clustered block name (clb_nlist.block_name[iblk]) and the block's top level pb name (clb_nlist.block_pb[iblk]->name). While these names both begin as the name of the cluster seed, pb names are modified during clustering (e.g. when adding a carry chain) and by the re-clustering API. 

When reading in a .net file, VPR populates clb_nlist.block_name() from the .net file (i.e. the top level pb name), and so when subsequently reading in a .place file the names may not match. VPR get around this by matching the .place file name against all atom names in the cluster. 

A while back I changed it so that VPR prints the top level pb name in the .place file instead of the block name, but I left the above matching code in place.

vaughnbetz, 2024-08-19:
Thanks @ZohairZaidi .  Yes, that update looks good to me.  If you make a pull request that would be great.

ZohairZaidi, 2024-08-19:
The pull request has been submitted, this issue can now be closed

vaughnbetz, 2024-08-20:
Thanks! It looks good and I'll merge it now.
If you're looking for another update: the placement file can now also (optionally) list a layer number.  We should really describe that in the documentation too. @amin1377 : could you point @ZohairZaidi at a sample 3D placement file?

amin1377, 2024-08-20:
Hi @ZohairZaidi,
I've attached a sample placement file. As you can see, it has six columns: block name, x, y, subblk, layer, and block number. For backward compatibility, we also accept a placement file that doesn't include the layer column. However, in placement files generated by newer versions of VPR, this column is always present, regardless of whether the FPGA architecture is 2D or 3D. The layer corresponds to the die number. In 3D FPGAs, multiple dies are stacked on top of each other, with the bottom die considered as layer 0, and the numbering continues upward. Please let me know if you have any questions.
[murax_stratixiv_arch_timing.txt](https://github.com/user-attachments/files/16675328/murax_stratixiv_arch_timing.txt)

amin1377, 2024-08-20:
One more thing, the suffix of the placement file is always `.place`. GitHub doesn't allow me to upload a file with that suffix, so I changed it to `.txt`.

ZohairZaidi, 2024-08-20:
Thanks for the explanation @amin1377. Really thrilled to gain all this knowledge. I just had a quick follow up question, if no layer_number is explicitly specified in the placement file, is the block assumed to be placed on the default layer which starts at 0 and goes upwards (e.g., layer 0, layer 1, layer 2, etc.)?

vaughnbetz, 2024-08-20:
Yes, with no layer we assume layer 0 (a single die system).

ZohairZaidi, 2024-08-21:
Very well, many thanks to you both for the help, I have submitted a new pull request. I updated the doc with the most recent file format and included an example of 3D place file alongside with one without the layer column. Please let me know if you spot any rookie mistakes by me :)

PR #2699 (2024-08-23): VPR Place File Format Doc Updated to latest changes

#### Description

VPR Place File Format Doc Updated to latest changes.

#### Related Issue

closes #2439

#### Motivation and Context

Up to date information in project docs

#### How Has This Been Tested?

#### Types of changes

- [ ] Bug fix (change which fixes an issue)
- [ ] New feature (change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to change)

#### Checklist:

- [ ] My change requires a change to the docum
closed 2024-08-23 by vaughnbetz