verilog-to-routing/vtr-verilog-to-routing #2451: Net max fanout value in Net Settings graphics control doesn't seem to work

vaughnbetz, 2023-11-28:
No matter what I set Net Max Fanout to in the Net Settings dialog box (tried 1, 20, 100), the display doesn't seem to change.

#### Expected Behaviour
Nets with a fanout above the specified threshold should not be drawn.

#### Current Behaviour
All nets appear to be drawn.

#### Steps to Reproduce
1. Run vpr with --disp on
2. When viewing the placement, click on Net Settings and set the Net max fanout to any value (e.g. a small one). The display doesn't change

#### Your Environment

* VTR revision: Nov. 15 master
* Operating System and version: Ubuntu
* Compiler version: g++

soheilshahrouz, 2023-12-09:
Fixed the issue and created a [PR](https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/2468).

soheilshahrouz, 2024-01-22:
Closing the issue as the PR was merged.

closed 2024-01-22 by soheilshahrouz