verilog-to-routing/vtr-verilog-to-routing #2463: Document how to test QoR for NoC designs

vaughnbetz, 2023-12-04:
#### Proposed Behaviour
The QoR section of the VTR developer guide should explain how to evaluate NoC quality metrics: what are they, what benchmark suites to run, what script to use to extract data etc. This probably requires iupdates and/or new subsections in two or three of the QoR evaluation sections 
[Key QoR Metrics](https://docs.verilogtorouting.org/en/latest/README.developers/#key-qor-metrics)
[Benchmark Selection](https://docs.verilogtorouting.org/en/latest/README.developers/#benchmark-selection)
[Collecting QoR Measurements](https://docs.verilogtorouting.org/en/latest/README.developers/#collecting-qor-measurements)

#### Current Behaviour
We don't say how to check NoC QoR

#### Context
We need to make NoC QoR measurement systematic so new developers can understand it.

soheilshahrouz, 2024-01-21:
Opened PR #2479

soheilshahrouz, 2024-02-03:
Closing the issue as the PR was merged.

closed 2024-02-03 by soheilshahrouz