verilog-to-routing/vtr-verilog-to-routing #2477: Error at parsing the results of a task

stef9998, 2024-01-12:
When running a task, vtr flow seems to run through normally. But then when it tries to parse the results there is an error about some assignment to a NoneType.

#### Expected Behaviour

Should just parse and not throw the error

#### Current Behaviour

Throws an error.

Example task-file: .../vtr_flow/tasks/test/config/config.txt

```
# Path to directory of circuits to use
circuits_dir=benchmarks/verilog

# Path to directory of architectures to use
archs_dir=arch/timing

# Add circuits to list to sweep
circuit_list_add=ch_intrinsics.v

# Add architectures to list to sweep
arch_list_add=k6_frac_N10_mem32K_40nm.xml

# Parse info and how to parse
parse_file=device_utilisation.txt

# Script parameters
script_params_list_add=--route_chan_width 80 -start odin
```
Example parse file: .../vtr_flow/parse/parse_config/device_utilisation.txt
```
# device utilisation

# Key values
device_utilisation;vpr.out;Device Utilization: (.*) \(target 1.00\)
```
standard-out:
```bash
.../vtr-working-directory$ $VTR_ROOT/vtr_flow/scripts/run_vtr_task.py test
k6_frac_N10_mem32K_40nm/ch_intrinsics		OK (took 1.69 seconds, overall memory peak 67.88 MiB consumed by vpr run)
Elapsed time: 1.76 seconds

Parsing test results...
Traceback (most recent call last):
  File ".../vtr-verilog-to-routing/vtr_flow/scripts/run_vtr_task.py", line 512, in <module>
    vtr_command_main(sys.argv[1:])
  File ".../vtr-verilog-to-routing/vtr_flow/scripts/run_vtr_task.py", line 280, in vtr_command_main
    num_failed = run_tasks(args, configs)
                 ^^^^^^^^^^^^^^^^^^^^^^^^
  File ".../vtr-verilog-to-routing/vtr_flow/scripts/run_vtr_task.py", line 328, in run_tasks
    parse_tasks(configs, jobs, args.alt_tasks_dir)
  File ".../vtr-verilog-to-routing/vtr_flow/scripts/python_libs/vtr/parse_vtr_task.py", line 195, in parse_tasks
    parse_task(config, config_jobs, alt_tasks_dir=alt_tasks_dir)
  File ".../vtr-verilog-to-routing/vtr_flow/scripts/python_libs/vtr/parse_vtr_task.py", line 214, in parse_task
    job.qor_parse_command()[0] = work_dir
    ~~~~~~~~~~~~~~~~~~~~~~~^^^
TypeError: 'NoneType' object does not support item assignment

```
When I don't run a taks but run vtr_flow and parse_vtr_flow seperate, it works fine:
```bash
.../vtr-working-directory$ $VTR_ROOT/vtr_flow/scripts/run_vtr_flow.py $VTR_ROOT/vtr_flow/benchmarks/verilog/ch_intrinsics.v $VTR_ROOT/vtr_flow/arch/timing/k6_frac_N10_mem32K_40nm.xml -temp_dir . --route_chan_width 80 -start odin
k6_frac_N10_mem32K_40nm/ch_intrinsics		OK (took 1.71 seconds, overall memory peak 67.94 MiB consumed by vpr run)
.../vtr-working-directory$ $VTR_ROOT/vtr_flow/scripts/python_libs/vtr/parse_vtr_flow.py . $VTR_ROOT/vtr_flow/parse/parse_config/device_utilisation.txt
device_utilisation	
0.69	
```

#### Possible Solution

#### Steps to Reproduce

Just running any task. I tried multiple and always get the same error.

#### Context

I was trying to ran a task for my bachelors thesis another student has created before.

#### Your Environment

* VTR revision used: github from around Dec.01.23
* Operating System and version: Arch Linux 
* Compiler version: Python 3.11.5 / GNU Make 4.4.1

soheilshahrouz, 2024-01-14:
run_vtr_task seems to expect a qor_parse_file field in the task configuration file. For a temporary fix, add qor_parse_file=qor_standard.txt to your task config file.

stef9998, 2024-01-14:
Hey, seems to work. I wrote myself a shell script the last days, so I probably won't use run_vtr_task to see if there will be some problems.

Just out of curiousity, why is this the case? The qor_parse_file seems to just add another parser.