verilog-to-routing/vtr-verilog-to-routing #2480: Problems in compiling the latest VTR

Outsider225, 2024-01-22:
#### Expected Behaviour

The compilation should be completed smoothly.

#### Current Behaviour

I encounter issues when making the yosys in vtr repository.
```
[ 66%] Built target libabc
Scanning dependencies of target abc
[ 66%] Building C object abc/CMakeFiles/abc.dir/src/base/main/main.c.o
[ 66%] Linking CXX executable abc
[ 66%] Built target abc
Scanning dependencies of target yosys
[ 67%] Generating yosys-bin
fatal: not a git repository: './/.git'
kernel/cellaigs.cc: In constructor ‘Yosys::Aig::Aig(Yosys::RTLIL::Cell*)’:
kernel/cellaigs.cc:395:39: error: call of overloaded ‘in(<brace-enclosed initializer list>)’ is ambiguous
  395 |   if (cell->type.in({ID($gt), ID($ge)}))
      |                                       ^
In file included from ./kernel/yosys.h:301,
                 from ./kernel/cellaigs.h:23,
                 from kernel/cellaigs.cc:20:
./kernel/rtlil.h:383:8: note: candidate: ‘bool Yosys::RTLIL::IdString::in(const string&) const’
  383 |   bool in(const std::string &rhs) const { return *this == rhs; }
      |        ^~
./kernel/rtlil.h:384:8: note: candidate: ‘bool Yosys::RTLIL::IdString::in(const Yosys::hashlib::pool<Yosys::RTLIL::IdString>&) const’
  384 |   bool in(const pool<IdString> &rhs) const { return rhs.count(*this) != 0; }
      |        ^~
kernel/cellaigs.cc:398:61: error: call of overloaded ‘in(<brace-enclosed initializer list>)’ is ambiguous
  398 |   int carry = mk.bool_node(!cell->type.in({ID($le), ID($ge)}));
      |                                                             ^
In file included from ./kernel/yosys.h:301,
                 from ./kernel/cellaigs.h:23,
                 from kernel/cellaigs.cc:20:
./kernel/rtlil.h:383:8: note: candidate: ‘bool Yosys::RTLIL::IdString::in(const string&) const’
  383 |   bool in(const std::string &rhs) const { return *this == rhs; }
      |        ^~
./kernel/rtlil.h:384:8: note: candidate: ‘bool Yosys::RTLIL::IdString::in(const Yosys::hashlib::pool<Yosys::RTLIL::IdString>&) const’
  384 |   bool in(const pool<IdString> &rhs) const { return rhs.count(*this) != 0; }
      |        ^~
make[4]: *** [kernel/cellaigs.o] Error 1
make[4]: *** Waiting for unfinished jobs....
kernel/satgen.cc: In member function ‘bool Yosys::SatGen::importCell(Yosys::RTLIL::Cell*, int)’:
kernel/satgen.cc:1268:32: warning: ‘undef_srst’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 1268 |      std::tie(d, undef_d) = mux(srst, undef_srst, rval, undef_rval, d, undef_d);
      |                             ~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
kernel/satgen.cc:1254:32: warning: ‘undef_ce’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 1254 |      std::tie(d, undef_d) = mux(ce, undef_ce, d, undef_d, old_q, undef_old_q);
      |                             ~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
kernel/satgen.cc:1240:32: warning: ‘undef_srst’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 1240 |      std::tie(d, undef_d) = mux(srst, undef_srst, rval, undef_rval, d, undef_d);
      |                             ~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[3]: *** [yosys/yosys-bin] Error 2
make[2]: *** [yosys/CMakeFiles/yosys.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [all] Error 2
``` 
Complete make log: 
[make.log](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/14004830/make.log)

#### Possible Solution

I noticed there is an issue https://github.com/YosysHQ/yosys/issues/3885 that reports the same problem, and the fix was merged on Aug 14, 2023. So I tried to replace the yosys folder of vtr with the yosys 0.37 cloned from yosysHQ [8649e30](https://github.com/YosysHQ/yosys/commit/8649e306682e44b7f3e1b869d1043041db00b13e), the compilation succeeds, and the regression tests pass as well. The yosys version in vtr is 0.32, released on Aug 7. Perhaps updating the yosys included with vtr could solve the issue.

#### Steps to Reproduce

1. `git clone --recurse-submodules https://github.com/verilog-to-routing/vtr-verilog-to-routing`
2. `cd vtr-verilog-to-routing`
3. `cd yosys`
4. `make config-gcc` because I don't have clang
5. `cd ..`
6. `make`

#### Your Environment

* VTR revision used: [97996a2](https://github.com/verilog-to-routing/vtr-verilog-to-routing/commit/97996a25cf5c50b1cb9f589cfb477279be7bbe4f)
* Operating System and version: Centos 7, CentOS Linux release 7.4.1708 (Core)
* Compiler version: gcc 9.3.1 20200408 (Red Hat 9.3.1-2)

Outsider225, 2024-01-22:
The yosys cloned from yosysHQ doesn't contain CMakeLists.txt, so I simply copied the CMakeLists.txt in the yosys folder of vtr.

`fatal: not a git repository: './/.git'` would appear when compiling the yosys of vtr, and this message also disappears after I replaced the yosys folder of vtr with the cloned yosys from yosysHQ.

poname, 2024-01-25:
https://docs.verilogtorouting.org/en/latest/quickstart/#environment-setup
>VTR requires several system packages and Python packages to build and run the flow. You can install the required system packages using the following command (this works on Ubuntu 18.04, 20.04 and 22.04, but you may require different packages on other Linux distributions). Our CI testing is on Ubuntu 22.04, so that is the best tested platform and recommended for development.

Outsider225, 2024-01-26:
Thanks for your suggestions. Now I'm using docker with Ubuntu 20.04, and VTR is functioning as expected.

closed 2024-01-26 by Outsider225