verilog-to-routing/vtr-verilog-to-routing #2489: Router Lookahead File Extension Error Handling

AlexandreSinger, 2024-02-12:
In previous commits of VPR, a file extension for the router lookahead such as ".bin" could be used without issue. In the most recent commit of VPR, an assertion is triggered when the extension of the router lookahead is anything but ".capnp".

It is not clear if this is intended behaviour; however, if it is a requirement to use the ".capnp" extension, then the error handling should be more clear than an assertion failure.

#### Expected Behaviour

If VPR is not supposed to accept files without the ".capnp" extension, it should provide an error at the beginning of execution and it should handle it more gracefully (as opposed to an assert statement).

If VPR is supposed to accept files without the ".capnp" extension, then the assert statement should probably be removed.

#### Current Behaviour

When passing in a router lookahead file using the "write_router_lookahead" option, the following error is produced when the file extension is ".bin".

```
/vpr/src/route/router_lookahead_map.cpp:466 write: Assertion 'vtr::check_file_name_extension(file_name, ".capnp")' failed.
```

AlexandreSinger, 2024-02-14:
Note that the VPR documentation does not specify which file extensions to use for router lookahead:
https://docs.verilogtorouting.org/en/latest/vpr/command_line_usage/#cmdoption-vpr-read_router_lookahead

However, the documentation for the RR graph is detailed as .xml for XML and .bin for Cap'N Proto:
https://docs.verilogtorouting.org/en/latest/vpr/command_line_usage/#cmdoption-vpr-read_router_lookahead

It should probably be a good idea to update the documentation with the expected file extensions as well. Also it would be less confusing if the RR graph file extension and the router lookahead extensions were consistent (.xml for XML and .capnp for Cap'N Proto; allowing .bin for backward compatibility).

amin1377, 2024-02-20:
This issue is addressed in PR #2492

closed 2024-03-14 by amin1377