verilog-to-routing/vtr-verilog-to-routing #2503: Add wire length attribute to RR graph output XML when using "--write_rr_graph" option

StephenMoreOSU, 2024-03-14:
On VTR master commit 9479f9d3685564a8ad97448f67bd99883fdcf6f9

When running the below command:

```bash
$VTR_ROOT/vpr/vpr \
    $VTR_ROOT/vtr_flow/arch/titan/stratixiv_arch.timing.xml \
    $VTR_ROOT/vtr_flow/benchmarks/blif/alu4.blif \
    --seed 55000 --route_chan_width 300 \
    --write_rr_graph rr_graph.xml \
```

The current output XML has the following for its segment fields: 

```xml
<segments>
<segment id="0" name="L4"><timing C_per_meter="1.80000007e-14" R_per_meter="201.699997"/>
</segment>
<segment id="1" name="L16"><timing C_per_meter="2.06999995e-14" R_per_meter="50.4199982"/>
</segment>
</segments>
```

I'd like this to be changed to the following output:

```xml
<segments>
<segment id="0" name="L4" length= 4><timing C_per_meter="1.80000007e-14" R_per_meter="201.699997"/>
</segment>
<segment id="1" name="L16" length= 16><timing C_per_meter="2.06999995e-14" R_per_meter="50.4199982"/>
</segment>
</segments>
```

This would then match the following description in the Stratix IV Architecture file:

```xml
<segment name="L4" freq="260" length="4" type="unidir" Rmetal="201.7" Cmetal="18.0e-15">
      <mux name="seg4_driver"/>
      <sb type="pattern">1 1 1 1 1</sb>
      <cb type="pattern">1 1 1 1</cb>
    </segment>
    <segment name="L16" freq="40" length="16" type="unidir" Rmetal="50.42" Cmetal="20.7e-15">
      <mux name="seg16_driver"/>
      <sb type="pattern">1 0 0 0 1 0 0 0 1 0 0 0 1 0 0 0 1</sb>
      <cb type="pattern">0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0</cb>
    </segment>
```

rs-dhow, 2024-06-11:
This is something I would use as well. It makes the initial portion of the routing graph output a more complete description of the architecture for routing purposes. Note that there is no RELIABLE way of deducing the intended (non-clipped) length of a segment over arbitrary array sizes. Thanks.

vaughnbetz, 2024-06-13:
Agreed, let's do this.  @amin1377 says he can get to it in the not-too-distant future.

rs-dhow, 2024-06-14:
That's great news. Thank you.

rs-dhow, 2024-08-14:
Thanks @amin1377. Your approach of always including it in output, and checking it in input if present, is the correct one. Current VPR behavior also means a CHANX/CHANY node should always have length(node) <= length(node's segment id), with equality when the node is not near the array edge and inequality when the node is clipped by the array edge since it otherwise would stick outside the edge. Thanks again.

StephenMoreOSU, 2024-08-14:
Thank you! @amin1377

closed 2024-08-14 by amin1377