verilog-to-routing/vtr-verilog-to-routing #2504: CI Test Failures on Master

AlexandreSinger, 2024-03-14:
Currently the CI is failing on Master without changes on the VTR side. The errors appear to all be related to ODIN II, since the CI tests that are failing are ODIN II regression tests.

See Test #8008

https://github.com/verilog-to-routing/vtr-verilog-to-routing/actions/runs/8273631785

Subsequent PRs are also failing with the same test failures.

#### S: Basic

This test fails when running the `vtr_reg_basic_odin` VTR task. I have run this task on my local development machine, and the test failures on the CI do not appear.

The failures appear to be a mix of ODIN II failures and VPR failures.

#### ODIN-II Basic Test

This test fails when running `odin_ii/regression_test/benchmark/suite/light_suite`.

There are a large number of test failures in this file. Some of them comparing a signed, 32-bit 0 to a 1-bit 0 and saying that is incorrect.

soheilshahrouz, 2024-03-15:
Apparently, this is a github actions [issue](https://github.com/actions/runner-images/issues/9491), and [the fix](https://github.com/actions/runner-images/pull/9513/files) has already been merged into master.

soheilshahrouz, 2024-03-16:
Closing the issue as the PR was merged.

closed 2024-03-16 by soheilshahrouz

AlexandreSinger, 2024-03-16:
Thank you so much Soheil!