verilog-to-routing/vtr-verilog-to-routing #2517: clang/LLVM-17 build

heshpdx, 2024-03-27:
Hello friends,
Can you check that LLVM-17 and LLVM-18 are are able to build VPR? @amin1377 and I are working on the SPEC CPU benchmark version of VPR and the next step is to show it works on a variety of systems and compilers. We are running into issues with the latest LLVM.

I've attached a log file below that shows our build issues. Although our source base is a little behind the mainline, I think the right thing to do is to fix the mainline and let that trickle down to us in SPEC CPU.

Majority of errors look like this:
```
In file included from vtr-vpr/libs/librrgraph/src/base/rr_graph_storage.cpp:2:
In file included from vtr-vpr/libs/libarchfpga/src/arch_types.h:11:
In file included from vtr-vpr/libs/libarchfpga/src/logic_types.h:14:
In file included from /spec/compilers/llvm-17.0.6/bin/../include/c++/v1/vector:321:
In file included from /spec/compilers/llvm-17.0.6/bin/../include/c++/v1/__format/formatter_bool.h:17:
In file included from /spec/compilers/llvm-17.0.6/bin/../include/c++/v1/__format/concepts.h:17:
In file included from /spec/compilers/llvm-17.0.6/bin/../include/c++/v1/__format/format_parse_context.h:16:
In file included from /spec/compilers/llvm-17.0.6/bin/../include/c++/v1/string_view:1059:
In file included from /spec/compilers/llvm-17.0.6/bin/../include/c++/v1/algorithm:1803:
In file included from /spec/compilers/llvm-17.0.6/bin/../include/c++/v1/__algorithm/nth_element.h:15:
/spec/compilers/llvm-17.0.6/bin/../include/c++/v1/__algorithm/sort.h:577:5: error: indirection requires pointer operand ('const edge_sort_iterator' invalid)

  577 |     *__begin = _Ops::__iter_move(__pivot_pos);
      |     ^~~~~~~~
 ```

[CPUv8.026.log](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/14775849/CPUv8.026.log)

Thank you!

AlexandreSinger, 2024-04-01:
Hi @heshpdx ,

I narrowed down the issue to this line in VPR: https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/5cfbaa0385169c4b218985fd34c2d3c99154b6e4/libs/librrgraph/src/base/rr_graph_storage.cpp#L158-L160

This is the member method of the `edge_sort_iterator` class that is passed into the std::sort method. The assumption made by the std::sort function is that the `first` and `last` parameters are "pointer-like". One property of pointers is that, when the pointer variable is made constant, the address it points to is constant but the value underneath may change; therefore the de-reference operator can be called on a `const int *` for example.

This `edge_sort_iterator` class does not have that property since its de-reference operator overload was not declared a constant method (I think by mistake). I think the std::sort method changed how it declares variables inside it as constant or not, which broke the VPR build.

I believe the solution to your problem would be to change this line in `/libs/librrgraph/src/base/rr_graph_storage.cpp`:
```
edge_swapper& operator*() const {
    return this->swapper_;
}
```
Notice that this method was made into a `const` method.

I can change this in the upstream version of VPR; however, I do not have the infrastructure to build and test on LLVM-17/18. 

Please let me know if this change resolves your issue!

AlexandreSinger, 2024-04-01:
@heshpdx I have raised a PR that should resolve this issue. The modification required was quite a bit more involved than I expected due to the member variable of the `edge_sort_iterator` not being a pointer. Changed it to a pointer in my PR; however, this may have an affect on performance. I do not think we can merge this into upstream until we verify that it does not have a negative affect on performance.

Let me know if the change resolves the issues on your end!

heshpdx, 2024-04-01:
Thank you! I will have the LLVM-17 tester try it out. You are right, it does have an impact on performance. I shared my findings in the PR.

heshpdx, 2024-04-02:
Update... even though the SPEC CPU version of VPR has had this llvm-17 build error for a long time, in the most recent integration we had replaced all instances of `std::sort` with `std::stable_sort` and updated all the input/output files based on the new results. So just yesterday, the latest report came back that showed llvm-17 is building on the reference machine (however it reports a verification error).

We are wondering if `std::stable_sort` has different and less stringent requirements than `std::sort` for LLVM's language implementation. That may be a better, less performance invasive method to fix this for everyone. Stay tuned while we dig in further. @amin1377

closed 2024-04-11 by heshpdx