verilog-to-routing/vtr-verilog-to-routing #2536: [Documentation] Missing Documentation on `--router_profiler_astar_fac`

AlexandreSinger, 2024-04-16:
While looking into how the CI runs the Titan benchmarks, specifically for `titan_quick_qor`:
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/3e8d605f292eea2c0a47964c21c56e995e76cad2/vtr_flow/tasks/regression_tests/vtr_reg_nightly_test2/titan_quick_qor/config/config.txt#L70C147-L70C174

I found that it sets an option called `--router_profiler_astar_fac` to 1.5. I noticed that no documentation on this command line argument exists on the VPR Command-Line Usage Documentation page:
https://docs.verilogtorouting.org/en/latest/vpr/command_line_usage/

Based on my quick investigation this appears to be used mainly in the placer, not necessarily in the router but I could be incorrect. Some documentation on what this option does, what its default value is, and how it is different from the astar_fac command-line option would be nice.

AlexandreSinger, 2024-04-17:
@amin1377 Can you handle this?

amin1377, 2024-04-17:
Hi Alex, thanks for filing the issue. Sure. I'll add the documentation.

amin1377, 2024-04-19:
I made the PR for that. However, we need to think about a way to automatically update the documents based on the parameters listed under `[VTR_ROOT]/vpr/base/read_options.cpp`

closed 2024-04-19 by amin1377

AlexandreSinger, 2024-04-19:
@amin1377 Thank you so much for handling this! I agree, perhaps some Github bot can be used which will detect when read_options.cpp has changed on master and automatically update the documentation if the version in master does not match. Something that can be investigated in the future!

AlexandreSinger, 2024-04-19:
Also, minor visual issue:
[image]

I assume you used a script to generate the documentation; why would the script not insert proper return carriages?
[image]

@amin1377

amin1377, 2024-04-29:
@AlexandreSinger Thanks, Alex. It is fixed now.

AlexandreSinger, 2024-04-29:
@amin1377 Thank you!