verilog-to-routing/vtr-verilog-to-routing #2548: VTR Quick Start Guide Inconsistencies

robluo, 2024-05-08:
While following the VTR quick start guide, I faced several issues that cannot be resolved by only reading the quick start guide's instructions. 
#### Issues encountered:

- [ ] `make env` did not work. Some issue with Python libraries
Setting the Python black library to 22.8.0 fixed the issue. (requirements.txt)
See: https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/2527/files
Wintermute cannot support 24.3.0 (its version of Ubuntu is too old).
Would require upgrading python
We would need to figure out what version of black to set this to if we want to fix this issue.
Preferably the minimum version to get this to work. However, 24.3.0 contains security updates we may want. Perhaps we should enforce an upgrade? -Alex

- [ ] The quickstart guide uses Odin as an example; however, Parmys is now the default tool. The quickstart should be updated to use Parmys or should say how to build VTR with Odin. The current quick fix is to build Odin by explicitly setting the flag in CMake. 
- [ ] Run_vtr_flow is having some issues.
Parmys is unable to parse the architecture files…
https://docs.verilogtorouting.org/en/latest/quickstart/#automatically-running-the-vtr-flow 
- [ ] In https://docs.verilogtorouting.org/en/latest/tutorials/flow/basic_flow/, the script parse_vtr_task.py is not needed since it is already part of the run_vtr_task.py script.

vaughnbetz, 2024-06-11:
@robluo : I think this was subsequently resolved. Can it be closed?

robluo, 2024-06-11:
@vaughnbetz For the second checkbox, @nedsels also raised an issue for it in #2551. For the third checkbox, it can be resolved by using run_vtr_task. I will close the issue.

closed 2024-06-11 by robluo