verilog-to-routing/vtr-verilog-to-routing #2551: Change ODIN II Quick Start Walkthrough to Parmys

nedsels, 2024-05-10:
We'd like to change the [Synthesizing with Odin II](https://docs.verilogtorouting.org/en/latest/quickstart/#synthesizing-with-odin-ii:~:text=targetted%20FPGA%20architecture.-,Synthesizing%20with%20ODIN%20II,%C2%B6,-First%20we%E2%80%99ll%20run) section in the Quick Start guide to use a Parmys example instead. @navidjafarof and @amin1377, how would we be able to do this?

#### Context

We want to change to Parmys since it's the default front-end, and move the current Odin example to the Odin tutorial page. However, I can't figure out how to run Parmys similarly to the Odin example. I've tried [installing Parmys](https://github.com/CAS-Atlantic/parmys-plugin#:~:text=make%20VTR_INSTALL_DIR%3D%60path%20to%20VTR%20build/install%20directory%60%20plugins%20%2Dj%60nproc%60), and get an error ``No rule to make target 'plugins'.``. When trying to [load Parmys in Yosys](https://github.com/CAS-Atlantic/parmys-plugin#:~:text=load%20the%20plugin-,plugin%20%2Di%20parmys,-%23%20read%20verilog), it says that it's not installed.

amirarjmand93, 2024-05-13:
@alirezazd please take a look at this issue.

vaughnbetz, 2024-06-11:
Nag, nag ... we should get this in.