verilog-to-routing/vtr-verilog-to-routing #2638: [CI] Nightly Test Failures

AlexandreSinger, 2024-07-02:
The last 5 scheduled nightly CI Test runs have failed due to QoR failures. I think some of the golden values are incorrect:
[image]

`vtr_reg_nightly_test1_odin`:
[image]

`vtr_reg_nightly_test2_odin`:
[image]

`vtr_reg_nightly_test3_odin`:
[image]

I think these three golden ranges just need to be adjusted.

AlexandreSinger, 2024-07-02:
@vaughnbetz Should we be concerned about these unexpected changes? Since these are not runtime issues these are things like negative slack and min_chan_width.

vaughnbetz, 2024-07-02:
No, I'm not very worried about them. Very small designs, just outside the QoR guardbands. Mostly Odin-II, which is also less important now.  I agree; I think somehow we didn't get the right golden values.  If you have the energy for it please go ahead and check in new golden results to fix them.

closed 2024-07-03 by AlexandreSinger