verilog-to-routing/vtr-verilog-to-routing #2660: Koios Benchmarks Failing With Flat Routing

nedsels, 2024-07-23:
With flat routing on, seemingly all ``koios_medium`` benchmarks fail. There are always some number of overused nodes.

#### Steps to Reproduce

I have created a new branch with a new regression test that is identical to ``regression_tests/vtr_reg_nightly_test4/koios_medium``, except it includes ``--flat_routing on`` as an argument.

1. Checkout [this branch](https://github.com/nedsels/vtr-verilog-to-routing/tree/koios_flat_routing).
2. Navigate to ``vtr_flow/tasks``
3. Run ``../scripts/run_vtr_task.py regression_tests/vtr_reg_nightly_test4/koios_medium_flat_routing``

``vpr`` should fail with exit code 2.

#### Your Environment

* VTR revision used: commit 3f2b9e8f2c4acc05cc3ab6884235fd4896777236
* Operating System and version: Ubuntu 18.04.6 LTS (wintermute)
* Compiler version: gcc version 9.4.0 (wintermute)

vaughnbetz, 2024-07-23:
This regtest doesn't seem to have the command line option to turn on choke point resolution.  Neither do other flat routing tests.  Shouldn't it be on?

script_params=-track_memory_usage -crit_path_router_iterations 100 --route_chan_width 300 --flat_routing on

vaughnb-cerebras, 2025-07-24:
@amin1377 : I think this is fixed, since we now turn on choke point resolution by default.  Amin can you check quickly?  We should probably run at least one koios design with flat routing on since it uses a different architecture, and hence provides some extra checking. @amin1377 : do we have such a test?