verilog-to-routing/vtr-verilog-to-routing #2677: VTR format checking / auto-format not checked by CI?

vaughnbetz, 2024-08-06:
Various PRs have shown formatting changes to break long multi-lines.  I think make format should do this automatically; this issue is to see if we have a missing rule in autoformat, or even worse, if auto-format is breaking manually made format changes for readability.

We should also have CI checking that our formatting is consistent.
Assigning to @soheilshahrouz since he has been cleaning up some code formatting; I think it would be better to figure out if there is a shortcoming in make format.  I think @jgoeders knows the autoformatting, so he could potentially help if you're stuck on how to improve it.

#### Expected Behaviour
make format should format long subroutine calls or subroutine variable declarations properly.  CI should test that we are format compliant.

#### Current Behaviour
I'm not 100% sure, but it seems long lines aren't caught by CI, and perhaps are not fixed by make format.

soheilshahrouz, 2024-10-28:
@vaughnbetz
Currently, CI doesn't check for long lines and `make format` does not break them into multiple shorter lines. We can set a line length limit in `.clang-format` file so that lines longer than the limit cause a CI failure. I think 80 is too short for widescreen displays. What do you thing about setting the line length limit to 120 characters?

vaughnbetz, 2024-10-28:
I'd be fine with 120 characters.

soheilshahrouz, 2025-04-09:
Closing this issue as it was resolved by merging #2935

closed 2025-04-09 by soheilshahrouz