verilog-to-routing/vtr-verilog-to-routing #2681: redundant adders when synthesizing addition with full adders

Junius00, 2024-08-09:
#### Current Behaviour

[image]

When using FPGA architecture files that use full adders in their CLBs which are carry-chained, synthesizing multi-bit addition tends to create redundant adders of the following:
- an adder before the adder chain that generates a dummy output at its `sumout`, and the `cout` is used as `cin = 1` for the first adder.
- an adder after the adder chain that generates a dummy output at its `cout`, and propagates the `cout` from the last adder to its `sumout`.

Here is very simple Verilog code for the above diagram generated by Yosys:
```
module test (
    input [3:0] a,
    output [4:0] c
);
    assign c = a + 4'b0001;
endmodule
```

#### Expected Behaviour

For the two cases above, I would expect to see:
- a constant `1` attached to the `cin` of the first adder (instead of a whole new adder).
- the `cout` of the last adder attached to an output bit (instead of being propagated through a whole new adder).

#### Steps to Reproduce
1. Use Yosys to synthesize the above Verilog code with:
    - `synthesis.tcl` under `vtr_flow/misc/yosys`
    - `k6_frac_N10_frac_chain_mem32K_40nm.xml` as the architecture file, under `parmys/parmys-plugin/tests/mips32r1_core`.

#### Context

I am currently attempting to perform architectural exploration to find the best architecture for unrolled DNN implementations. This unrolling means the hardening of weights into many of these adder chains, which quickly leads to errors due to this issue because:
- The number of adders to pack has increased by ~1.5x.
- These redundant adders also add their dangling dummy outputs which are not swept (since they are being driven), which leads to the `Can not find any logic block that can implement molecule`, with the molecule mentioned being the dummy output.

#### Your Environment
* VTR revision used: v8.0.0
* Operating System and version: Ubuntu 22.04.4 LTS

rs-dhow, 2024-08-09:
Some comments.

1. If you are seeing ~1.5x increase in adder bits, that would suggest you have a lot of 4b adders, to which cin-generator and cout-router adders are added. The overhead is much smaller of course with larger carry chains.
2. In most archs I've looked at, the cout --> cin connection is not routable. That means you cannot rely on CHAN/IPIN nodes to get in the desired '0' or '1' constant to the cin of the LSB of a carry chain. The solution is to add an extra adder, as you note, below the LSB. This is the only way to make cin controllable when it's not routable and has no built-in mux. This happens in almost all user carry chains.
3. The same comment applies to making the final cout observable: you need to add an extra adder beyond the MSB through which cout will flow to its routable output. The frequency of this depends on the design and how it uses adders.

The tradeoff here is whether you want to add a mux on cin, slowing down all carry chains, and/or a cout-driven extra input to an output mux driving the routable network, which slows down all non-carry logic also flowing through that mux. In industry, they usually do the latter, and the former is done only for certain bits (which creates an alignment/positioning issue in adder packing/placement).

Concerning the error message you mention in the second bullet under "Context", this must be from an error in your vpr.xml. We have both kinds of extra adders and do not see this message.

Good luck.

Junius00, 2024-08-09:
Appreciate your prompt response and insightful explanation and suggestions.

closed 2024-08-09 by Junius00