verilog-to-routing/vtr-verilog-to-routing #2684: Placement Constraints not working

GopalN, 2024-08-13:
I am trying to place a simple AND gate primitive in the CLB of an architectural file. I generated the constraints file as given in the documentation. 
a and b is input c is the output
<vpr_constraints tool_name="vpr">
	<partition_list>
		<partition name="c">
			<add_atom name_pattern="c"/>
			<add_region subtile="0" x_high="2" x_low="2" y_high="6" y_low="6"/>
		</partition>
		<partition name="out:c">
			<add_atom name_pattern="out:c"/>
			<add_region subtile="0" x_high="7" x_low="7" y_high="13" y_low="13"/>
		</partition>
		<partition name="a">
			<add_atom name_pattern="a"/>
			<add_region subtile="0" x_high="7" x_low="7" y_high="0" y_low="0"/>
		</partition>
		<partition name="b">
			<add_atom name_pattern="b"/>
			<add_region subtile="0" x_high="8" x_low="8" y_high="0" y_low="0"/>
		</partition>
<	/partition_list>
</vpr_constraints>
#### Expected Behaviour

The expected behaviour is according to the constraint file it should get placed at the CLB location 2,6.
#### Current Behaviour

There is noo change in the placement and it gets placed in the 7,7 CLB tile .
This is part of the relevant log output

Timing analysis: ON
Circuit netlist file: and2.net
Circuit placement file: and2.place
Circuit routing file: and2.route
Circuit SDC file: and2.sdc
Vpr floorplanning constraints file: ../../../../arch/vpr_constraints2.xml

PlacerOpts.place_freq: PLACE_ONCE
PlacerOpts.place_algorithm: CRITICALITY_TIMING_PLACE
PlacerOpts.pad_loc_type: FREE
**PlacerOpts.constraints_file: No constraints file given**
PlacerOpts.place_cost_exp: 1.000000
PlacerOpts.place_chan_width: 40
PlacerOpts.inner_loop_recompute_divider: 0
PlacerOpts.recompute_crit_iter: 1
PlacerOpts.timing_tradeoff: 0.500000
PlacerOpts.td_place_exp_first: 1.000000
PlacerOpts.td_place_exp_last: 8.000000
PlacerOpts.delay_offset: 0.000000
PlacerOpts.delay_ramp_delta_threshold: -1
PlacerOpts.delay_ramp_slope: 0.000000
PlacerOpts.tsu_rel_margin: 1.000000
PlacerOpts.tsu_abs_margin: 0.000000
PlacerOpts.post_place_timing_report_file: 
PlacerOpts.allowed_tiles_for_delay_model: 
PlacerOpts.delay_model_reducer: MIN
PlacerOpts.delay_model_type: DELTA
PlacerOpts.rlim_escape_fraction: 0.000000
PlacerOpts.move_stats_file: 
PlacerOpts.placement_saves_per_temperature: 0
PlacerOpts.effort_scaling: CIRCUIT
PlacerOpts.place_delta_delay_matrix_calculation_method: ASTAR_ROUTE
PlaceOpts.seed: 1
AnnealSched.type: AUTO_SCHED
AnnealSched.inner_num: 0.500000

#### Your Environment

VPR FPGA Placement and Routing.
Version: 8.1.0-dev+v8.0.0-10571-g6e8ac62b7
Revision: v8.0.0-10571-g6e8ac62b7
Compiled: 2024-06-11T11:33:13
Compiler: GNU 11.4.0 on Linux-6.5.0-35-generic x86_64
Build Info: release IPO VTR_ASSERT_LEVEL=2

soheilshahrouz, 2024-08-13:
Thanks for submitting this issue.
Can you please share the architecture and netlist files (with .xml and .blif formats) you passed to VPR?

GopalN, 2024-08-14:
Thank you very much for your response.
I am working on the SOFA epfga architecture with 14* 14 blocks .I have attached the XML file and the blif file of the FPGA .I am trying to implement a simple and gate in particular location .
[files.zip](https://github.com/user-attachments/files/16609242/files.zip)

soheilshahrouz, 2024-08-14:
Your constraints file is not compliant with [expected file format](https://docs.verilogtorouting.org/en/latest/vpr/placement_constraints/). It is missing `<partition>` tags under `<partition_list>`. Here is the constraint file I used:

```xml
<vpr_constraints tool_name="vpr">
    <partition_list>

        <partition name="Part0">
            <add_atom name_pattern="c"/>
            <add_region subtile="0" x_high="2" x_low="2" y_high="6" y_low="6"/>
        </partition>

        <partition name="Part1">
            <add_atom name_pattern="out:c"/>
            <add_region subtile="0" x_high="7" x_low="7" y_high="13" y_low="13"/>
        </partition>

        <partition name="Part2">
            <add_atom name_pattern="a"/>
            <add_region subtile="0" x_high="7" x_low="7" y_high="0" y_low="0"/>
        </partition>

        <partition name="Part3">
            <add_atom name_pattern="b"/>
            <add_region subtile="0" x_high="8" x_low="8" y_high="0" y_low="0"/>
        </partition>

    </partition_list>
</vpr_constraints>

```

I ran the most recent commit (21d015036969a955298cbc9b94ca74fd35a016be) on master branch with the given architecture and netlist files and the placement engine adhered to the constraints. I needed to remove a few tileable rr graph related attributes in the architecture file as they are only supported in the openfpga branch.

vaughnbetz, 2024-08-15:
You also have to tell vpr to use the placement constraints file at the command line. See https://docs.verilogtorouting.org/en/latest/vpr/command_line_usage/#cmdoption-vpr-read_vpr_constraints

GopalN, 2024-08-16:
@vaughnbetz 
As you can see from the log file I have assigned the constraints file on the command line and which it is recognising .

Timing analysis: ON
Circuit netlist file: and2.net
Circuit placement file: and2.place
Circuit routing file: and2.route
Circuit SDC file: and2.sdc
Vpr floorplanning constraints file: ../../../../arch/vpr_constraints2.xml

Also I tried the constraint file given by @soheilshahrouz still it is not working. Can you also share what changes you made in the architectural file.

GopalN, 2024-08-16:
Solved it .When i remove the --analysis --disp on from the command line it works .Is it a known bug or am I missing something.

vaughnbetz, 2024-08-16:
--analysis means you just want to look at the results of a prior compile.  So that's not a bug; you didn't run the placement engine with the constraint file.

GopalN, 2024-08-17:
Thank you Sir for the reply and support.
I realised this later .
I am closing this comment

closed 2024-08-17 by GopalN