verilog-to-routing/vtr-verilog-to-routing #2697: ./kernel/yosys.h:42:10: fatal error: 'map' file not found

Martoni, 2024-08-18:
Following BUILDING.md, got a Yosys compile error.

#### Expected Behaviour

No yosys compile error

#### Current Behaviour

```bash
Building target(s): 
[ 63%] Built target libabc
[ 63%] Built target abc
[ 65%] Generating yosys-bin
In file included from kernel/driver.cc:20:
./kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
1 error generated.
make[4]: *** [Makefile:759 : kernel/driver.o] Erreur 1
make[3]: *** [yosys/CMakeFiles/yosys.dir/build.make:73 : yosys/yosys-bin] Erreur 2
make[2]: *** [CMakeFiles/Makefile2:1153 : yosys/CMakeFiles/yosys.dir/all] Erreur 2
make[1]: *** [Makefile:146 : all] Erreur 2
make: *** [Makefile:94 : all] Erreur 2

```

#### Steps to Reproduce

Following BUILDING.md :

```bash
$ git submodule init
$ git submodule update
$ ./install_apt_packages.sh
$ sudo apt-get install libc++-dev # added for test
$ make env
$ source .venv/bin/activate
$ pip install -r requirements.txt
$ make
```

#### Your Environment

* VTR revision used: master (f13f87b5aa74e9c6fa9c13c3466e2ce82003fb41)
* Operating System and version: 
```$ cat /etc/lsb-release 
DISTRIB_ID=LinuxMint
DISTRIB_RELEASE=21
DISTRIB_CODENAME=vanessa
DISTRIB_DESCRIPTION="Linux Mint 21 Vanessa"
```
* Compiler version: Don't know

AlexandreSinger, 2024-08-19:
Hi @Martoni , thanks for the issue!

This was actually a known issue on Yosys' side which was resolved recently:
See: https://github.com/YosysHQ/yosys/issues/681 and https://github.com/YosysHQ/yosys/pull/4234

VTR uses a slightly older version of Yosys which does not have this fix yet. We are currently in the process of upgrading Yosys in VTR up to the most recent version: See PR #2646 

In summary, the issue is caused by Yosys implicitly setting `CXX` to be `gcc` and `clang` instead of `g++` and `clang++`; this causes it to be unable to include the C++ standard library. This issue only manifests itself on OS's that are not Ubuntu (for some reason I am not completely sure). Here in VTR, we mainly develop on Ubuntu; which is the operating system that we recommend.

As a quick fix, you can resolve the issue in Yosys' makefile here (locally) by setting CXX to be proper: https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/yosys/Makefile

However, if you have access to a machine with a recent version of Ubuntu installed, I recommend using that instead for a smoother installation.

Martoni, 2024-08-19:
Thank you very much for the detailed explanation.
I'll try it this evening.

Martoni, 2024-08-19:
Ok, works with changing clang to gcc in yosys/Makefile
Thanks

closed 2024-08-19 by Martoni