verilog-to-routing/vtr-verilog-to-routing #2704: [IPA] MVP to manually test vpr --server mode

w0lek, 2024-08-26:
The vpr contains --server mode functionality. 
In order to check that feature we actually requires the separate client application.

#### Proposed Behaviour

New MVP IPAClient to work in conjunction with vpr --server mode.

w0lek, 2024-08-26:
The MVP application is ready as separate github repo.
https://github.com/w0lek/IPAClient

The usage https://github.com/w0lek/IPAClient/blob/main/README.md

Things to discuss/implement:
1) how IPAClient would be integrated into vtr/vpr
2) documentation/reference from vtr/vpr doc to IPAClient

w0lek, 2024-08-26:
@vaughnbetz 
i created ticket to track the progress about IPAClient progress.

Things to discuss/implement:
1) how IPAClient repo would be integrated into vtr/vpr repo.
2) documentation/reference from vtr/vpr doc to IPAClient subproject.

w0lek, 2024-09-12:
the PR is created
https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/2719

vaughnbetz, 2025-09-13:
Closing as this was done.