verilog-to-routing/vtr-verilog-to-routing #2705: [Build] VTR Fails to Build on ASSERT_LEVEL=4

AlexandreSinger, 2024-08-27:
The current version of VTR (master) does not build with `-DVTR_ASSERT_LEVEL=4`. We are getting a few errors, but there may be more:
[image]
I ran into this while trying to debug some code elsewhere in VTR.

According to the VTR documentation: https://docs.verilogtorouting.org/en/latest/api/vtrutil/logging/#vtr-assertion

```
 VTR_ASSERT_LEVEL == 4: VTR_ASSERT_OPT, VTR_ASSERT, VTR_ASSERT_SAFE, VTR_ASSERT_DEBUG enabled
 VTR_ASSERT_LEVEL == 3: VTR_ASSERT_OPT, VTR_ASSERT, VTR_ASSERT_SAFE enabled
 VTR_ASSERT_LEVEL == 2: VTR_ASSERT_OPT, VTR_ASSERT enabled
 VTR_ASSERT_LEVEL == 1: VTR_ASSERT_OPT enabled
 VTR_ASSERT_LEVEL == 0: No assertion checking enabled
```
VTR_ASSERT_LEVEL == 4 will turn on the "ASSERT_DEBUG" statements in the code.

Looking through the CI; it looks like the CI only tests up to VTR_ASSERT_LEVEL 3. This makes sense since the debug asserts may make things slow; but it defeats the point if the debugging code in VTR has bugs.

These errors should be fixed and a CI test should be added to guard against this regressing. The test should at least build VTR and (if it does not take too long) should run the basic or preferably the strong tests so every debug assert is hit.

soheilshahrouz, 2024-08-28:
It looks like PR #2669 introduced this compilation error. I'll try to fix it quickly.

AlexandreSinger, 2024-08-28:
Thanks @soheilshahrouz !

closed 2024-09-16 by AlexandreSinger