verilog-to-routing/vtr-verilog-to-routing #2722: Improve 3D switch block commenting and move command line option to arch file

vaughnbetz, 2024-09-13:
PR #2370 has some suggested commenting updates; please take a look and integrate them as we merged that PR so it wasn't hanging around too long.

#### Proposed Behaviour

- [ ] Move --custom_3d_sb_fanin_fanout to the arch file instead. Not sure what section is best ... maybe in the custom switch block or the switch block list is the most logial.
- [ ] Add a big comment somewhere saying how custom_3d_sb_fanin_fanout works. It should detail exactly what the rr graph this generates looks like.
- [ ] Add a comment describing why you are checking if you found nodes on the target layer in the high fanout route expansion https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/2370/files/b681f8d4afc16ab837c5a03cbfa353a17098be37#diff-94aa757a7f9f207973239bdae6ecbbdd7a8c7e916d4d6f3ce3a349adc849d6c5R1022

vaughnbetz, 2025-09-14:
@soheilshahrouz : I think this will be handled by your pending updates. Sending this to you to close or comment on when those updates are ready. ￼

soheilshahrouz, 2026-09-15:
With scatter-gather merged into master, I think we can close this issue.

closed 2026-09-15 by soheilshahrouz

vaughnbetz, 2026-09-15:
Yes, agreed!