verilog-to-routing/vtr-verilog-to-routing #2739: Update VTR quick start guide to use yosys / parmys

vaughnbetz, 2024-09-21:
The VTR quick start guide uses Odin II as the synthesis tool, but the yosys/parmys flow is now the default VTR flow. We should update the quick start to use yosys/parmys.  The Odin II section should be moved down to a later, optional part of the quick start that just describe how to run Odin II (as the primary example of yosys/parmys would have already covered the whole flow).

#### Proposed Behaviour
Change all mention of Odin II to the equivalent parmys / yosys flow commands.  Move the Odin II command(s) down to the end of the quick start under an Odin II section.

#### Current Behaviour
Odin II is the only synthesis tool described in the quick start.

vaughnbetz, 2024-09-21:
@ZohairZaidi : this would be a great issue to fix if you have the time ... it's a bit annoying that the quick start doesn't cover our newly default synthesis flow.

ZohairZaidi, 2024-09-25:
Hi @vaughnbetz, just to clarify, you are referring to the "Manually Running the VTR Flow" section right?

vaughnbetz, 2024-09-25:
Yes, that's right.

ZohairZaidi, 2024-10-11:
Hi @vaughnbetz, I was wondering if you knew the equivalent steps to generate the blif file outside of the VTR flow script as I was trying to follow the guide posted on the original parmys GitHub repo (https://github.com/CAS-Atlantic/parmys-plugin) but it seems to be out of date.

vaughnbetz, 2024-10-11:
I haven't done it, so I'm not sure. @alirezazd @amirarjmand93 : can you summarize the commands?  @ZohairZaidi : you can also probably infer them by printing out / watching the various commands run by run_vtr_flow.

ZohairZaidi, 2024-10-11:
I just realized that my issue has been answered in the original repo. thanks for the help.

PR #2776 (2024-11-18): [QuickStart] Yosys/Parmys Update

#### Description

This pull request updates the VTR Quick Start Guide by replacing Odin II as the primary synthesis tool with the Yosys/Parmys flow. I have also updated the graphic images
#### Related Issue

closes #2739  

#### Checklist:

- [x] My change requires a change to the documentation
- [x] I have updated the documentation accordingly
- [ ] I have added tests to cover my changes
- [ ] All new and existing tests passed
closed 2024-10-16 by ZohairZaidi