verilog-to-routing/vtr-verilog-to-routing #2750: Turn on choke point analysis in the router by default, and make help etc. more user-friendly

vaughnbetz, 2024-09-27:
Many architectures need choke point analysis to be on to converge when flat routing is selected. Right now users have to turn this on themselves, which makes flat routing harder to use. We should turn this option on by default (only flat routing will be affected) and we should improve its help and add it to the command line documentation.

#### Proposed Behaviour
route_grp.add_argument(args.has_choking_spot, “–has_choking_spot”)
Rename to –router_opt_choke_points
New help: Some FPGA architectures with limited fan-out options within a cluster (e.g. fracturable LUTs with shared pins) do not converge well in routing unless these fan-out choke points are discovered and optimized for during net routing. This option helps router convergence for such architectures.
Default: On (turns off for non-flat) (measure downside)
Other values on / off.

#### Current Behaviour
Not on by default, and the name isn't great. Also not in the VTR documentation.  It does have a CPU time hit: 11% total VTR time and 27% memory footprint on Titan to turn this on for the SIV architecture (which doesn't need it).  But now the flagship architecture will route by default.

#### Possible Solution

- [ ] Rename to --router_opt_choke_points
- [ ] Turn on by default
- [ ] Update help to "Some FPGA architectures with limited fan-out options within a cluster (e.g. fracturable LUTs with shared pins) do not converge well in routing unless these fan-out choke points are discovered and optimized for during net routing. This option helps router convergence for such architectures. Turning it off when you know an architecture has no choke points will save cpu time with flat routing as the necessary data structures will not be built."
- [ ] Add to VTR documentation.
- [ ] Try to optimize the memory footprint and run time (not gating; can merge the other items first)

#### Context
QuickLogic (and others) have found flat routing doesn't converge when they first run it. This will make it easier to use.

@duck2 @ganeshgore

fkosar-ql, 2024-09-27:
Just checked with vtr_reg_strong and setting choke point analysis to true seems to fix all routing failures with the flat router

vaughnb-cerebras, 2024-09-27:
Yaay!  Amin is going to turn it on by default.

vaughnbetz, 2025-06-13:
@amin1377 : can you check if this is done?  If it is, let's close this.
It is on by default, but I think the docs are not there.