verilog-to-routing/vtr-verilog-to-routing #2753: VPR Test `test_odd_even_routing` Failed

ueqri, 2024-09-27:
The VPR test `test_odd_even_routing` (as shown below) has consistently failed in CI (U: C++ Unit Tests) since https://github.com/verilog-to-routing/vtr-verilog-to-routing/commit/2b6a935494363f8be1a83a05253c9bd70106e400 (included) in the master branch.

https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/3e53a937f772bcb81998e57ae074e2a929d49420/vpr/test/test_odd_even_routing.cpp#L228-L264

@AlexandreSinger @soheilshahrouz It seems that Alex's commit https://github.com/verilog-to-routing/vtr-verilog-to-routing/commit/2b6a935494363f8be1a83a05253c9bd70106e400 has nothing to do with either NoC (or odd even routing). I am not sure why this test started failing since then.

The corresponding logs are here:

```text
-------------------------------------------------------------------------------
test_route_flow
  Test case where multiple traffic flows are router, and routes are checked for
  turn legality and deadlock freedom.
-------------------------------------------------------------------------------
/home/runner/work/vtr-verilog-to-routing/vtr-verilog-to-routing/vpr/test/test_odd_even_routing.cpp:228
...............................................................................

/home/runner/work/vtr-verilog-to-routing/vtr-verilog-to-routing/vpr/test/test_odd_even_routing.cpp:228: FAILED:
due to a fatal error condition:
  SIGSEGV - Segmentation violation signal
```

Showed in https://github.com/verilog-to-routing/vtr-verilog-to-routing/actions/runs/11077555018/job/30783077163?pr=2720.

```text
-------------------------------------------------------------------------------
test_route_flow
  Test case where multiple traffic flows are router, and routes are checked for
  turn legality and deadlock freedom.
-------------------------------------------------------------------------------
/home/runner/work/vtr-verilog-to-routing/vtr-verilog-to-routing/vpr/test/test_odd_even_routing.cpp:228
...............................................................................

/home/runner/work/vtr-verilog-to-routing/vtr-verilog-to-routing/vpr/test/test_odd_even_routing.cpp:228: FAILED:
due to unexpected exception with message:
  std::bad_alloc
```

Showed in https://github.com/verilog-to-routing/vtr-verilog-to-routing/actions/runs/11072279016/job/30766140332 and same in https://github.com/verilog-to-routing/vtr-verilog-to-routing/actions/runs/11070489508/job/30760275852.

AlexandreSinger, 2024-09-28:
@ueqri Thanks for creating an issue on this. Soheil and I have been informally talking about this since Monday. It started since around when I made the ClusterLegalizer change. @soheilshahrouz has been looking into it, but is saying that he cannot replicate it locally. It seems to be something with the exact environment of the CI which is causing this test to randomly fail. I am not familiar with this Unit Test to know why.

ueqri, 2024-09-28:
Gotcha! Thanks, Alex and Soheil! I was just trying to make CI happy in the router PR and found this error that I have no idea how to resolve. 😂

AlexandreSinger, 2024-09-28:
Yeah, I am thinking of disabling this test while we figure this out; but you can just try to rerun the test.

soheilshahrouz, 2024-11-03:
Resolved by mergin #2735.

closed 2024-11-03 by soheilshahrouz