verilog-to-routing/vtr-verilog-to-routing #2754: [Routing] Incremental Slack Timing Analysis Does Not Match When Computed From Scratch

AlexandreSinger, 2024-09-29:
While working on the Analytical Placer, I generated a very basic clustering and cluster placement. The quality of both of these was meant to be low as this is mainly just a PoC flow through VPR. These tests go through routing without an issue; however, for one test they fail.

When the assert level is set to 4, there is a debug check which verifies that the incremental slack timing analysis gets the same answer as one computed from scratch:
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/3e53a937f772bcb81998e57ae074e2a929d49420/vpr/src/timing/slack_evaluation.cpp#L286-L288

I have found on my testcase, this check gets tripped.

VTR was built with the following command:
```
make -j12 CMAKE_PARAMS="-DWITH_ODIN=off -DVTR_ASSERT_LEVEL=4"
```
Note: The assert level being 4 is important.

The testcase circuits and a testing script can be found in this tar file:
[timing_bug.tar.gz](https://github.com/user-attachments/files/17179821/timing_bug.tar.gz)

It contains a bash script with the command I used to run VPR:
```sh
VTR_DIR=~/vtr-verilog-to-routing                                                                                                                                                                            
                                                                                                                                                                                                            
$VTR_DIR/vpr/vpr \                                                                                                                                                                                          
    fixed_k6_frac_N8_22nm.xml \                                                                                                                                                                             
    ch_intrinsics \                                                                                                                                                                                         
    --circuit_file ch_intrinsics.pre-vpr.blif \                                                                                                                                                             
    --net_file ch_intrinsics.net \                                                                                                                                                                          
    --place_file ch_intrinsics.place \                                                                                                                                                                      
    --route \                                                                                                                                                                                               
    --device unnamed_device \                                                                                                                                                                               
    --read_vpr_constraints ch_intrinsics_fixed_io.xml \                                                                                                                                                     
    --min_route_chan_width_hint 40
```
The output looks like as follows:
[image]

This is the second routing attempt after routing at a channel width of 40.

Interestingly, if you do not provide the min chan width hint of `--min_route_chan_width_hint 40` this does not happen.

AlexandreSinger, 2024-09-29:
Refer to PR #2752 for more context.

AlexandreSinger, 2024-10-12:
This issue is now appearing in Master. Needs to be investigated soon.
https://github.com/verilog-to-routing/vtr-verilog-to-routing/actions/runs/11301269116/job/31435439924

AlexandreSinger, 2024-10-15:
@vaughnbetz I found the problem. It is not a precision issue it is a bit more nuanced than that. Investigating the code, we actually want the floating point values to be exactly equal in this case since the slack is not being recomputed, but is instead finding the min of, which does necessarily need to be equal.

The issue is that the incremental timing analysis is **not stably deterministic**. When there is a tie for the lowest slack, the first node checked by the analysis is used. If the order in which you check the nodes is different you will get a different worst slack node (but the same worst slack). This is why the errors always say that the "Calculated worst slack nodes does not match", but the worst slack always does match.

To fix the issue, in the incremental and recompute slack functions, I added a tie-breaking condition which would prefer the node with the lower ID in the case of a tie for minimum slack (similar to what we did for the parallel router). This also should be fixed for the required times as well as the slack since they are computed in the same way. What do you think about this change?

vaughnbetz, 2024-10-15:
As we discussed in the hall, I think we can simply stop checking which nodes are returned, and instead just check that the worst case slacks are equal.  This case occurs when we have multiple critical paths, and it isn't really a problem; we base all optimization decisions on slack, not which node is the most critical in the whole circuit.

rs-dhow, 2024-10-15:
Your decision here is fine, but I would point out that this is a case of some related issues I've seen in EDA since Actel in the 80s. Namely, (i) it's best if comparisons are complete: if two objects do not refer to the same memory region, they cannot be equal (your ID tie-breaker achieves this); (ii) sorts should be stable; (iii) enumeration order should never depend on memory addresses (nothing should depend on memory addresses). FWIW YMMV etc. Enjoy.

PR #2774 (2024-10-17): [STA] Fixed Incremental Slack Timing Analysis Verify Code
In debug mode, the STA checks if the incremental STA matches the STA computed from scratch. In this check it checked for if the critical nodes matched; however, the nodes returned are not stably deterministic (it does not break ties in a deterministic way). After discussing with Vaughn, it is safe to just remove the nodes check since it would not change the result of the analysis.

closes #2754
commit e45f0cd0 (2024-10-17): [STA] Fixed Incremental Slack Timing Analysis Verify Code

In debug mode, the STA checks if the incremental STA matches the STA
computed from scratch. In this check it checked for if the critical
nodes matched; however, the nodes returned are not stably deterministic
(it does not break ties in a deterministic way). After discussing with
Vaughn, it is safe to just remove the nodes check since it would not
change the result of the analysis.

closes #2754