verilog-to-routing/vtr-verilog-to-routing #2767: Extend XY_SPECIFIED syntax support to fixed layout XML files

petergrossmann21, 2024-10-11:
The recent pull request for XY_SPECIFIED custom switch block specification https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/2664 was tested with an architecture with an <auto_layout> tag.

In the use case that I have for this feature the requirement will be to use it on layouts using the <fixed_layout> tag instead.  Testing on an architecture of this type yielded the following error

`Error 1: /projects/zeroasic/fpgaa_working/upgrade_custom_switchblock_xml/fpga_architect/test/fpga_gamma_vanilla_baby_falcon/cad/fpga_gamma_vanilla_baby_falcon.xml:489 Specifying SB locations for auto layout devices are not supported yet!`

It would appear that the error message is checking for the wrong layout type (or reporting the wrong layout type).

@saaramahmoudi , is the feature expected to work for both layout types?  If so, we can pick some fixed layout architecture and add an additional test case as part of closing the error message.

In the event that I'm the only user of the feature in this use case I may be able to contribute to the development of a fix, as long as we are lenient with respect to schedule.

saaramahmoudi, 2024-10-15:
Hi @petergrossmann21, 

The feature is not supported for auto_layout yet, but the process is very easy. When we are parsing the custom switchblock specification in read_xml_arch.cpp, I don't know what it the grid_width and grid_height if we are using auto_layout feature (since that is decided in packer step not in the parsing step). The solution would be to save these information from architecture file as a string in read_xml_arch.cpp and perform the actual parsing in rr_graph.cpp when the packer has already finished, meaning that we just need to move my code from one place to another. 
I was planning to do that with my next PR. However, I am a bit busy right now and can't do that immediately. 
Hope that helps and let me know if you have more questions.

saaramahmoudi, 2024-10-15:
@petergrossmann21 I used one of the existing architecture to create the task, however I passed --device in task config file if you look into this file: [strong_custom_sb_loc](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vtr_flow/tasks/regression_tests/vtr_reg_strong/strong_custom_sb_loc/config/config.txt)

petergrossmann21, 2024-10-15:
Update:  I am able to reproduce the error message described above with the provided vtr_flow/arch/custom_sb_location/stratixiv_arch.timing.xml

petergrossmann21, 2024-10-15:
Aha, right you are -- I see the fixed_layout tag below now in your example.  This is a feature I was not familiar with.

My architecture contains only a fixed_layout tag; let me play around with --device and see what I turn up.

@vaughnbetz What are your thoughts on how features that require <fixed_layout> should be supported?  Should it be possible for a feature of this type to work without use of the --device option at the CLI?

petergrossmann21, 2024-10-15:
Update:  adding a --device option to the VPR command line when running my own architecture works.  It appears that the item to close is the matter of whether adding this command-line option should be required to use this feature.

vaughnbetz, 2024-10-17:
Adding @AlexandreSinger since he was also wondering about how he'll handle multiple fixed devices in his testing.  I think specifying a --device on the command line makes sense.  We default to auto if it isn't specified, which seems reasonable. The only other default I can think of would be : the only layout in the arch (if there is only one), otherwise auto.  Not sure if that would be more clear or more confusing though.

petergrossmann21, 2024-10-17:
With the bias that I exclusively work with architecture files for which there is only one layout in the arch and it is of type <fixed_layout>, I find the "only layout" else auto default appealing.  However, I'm not sure that it's really worth setting it up that way if it causes headaches.  Happy to discuss further in a call (weekly sync-up or schedule a smaller discussion)

AlexandreSinger, 2024-10-17:
@vaughnbetz I spent the last couple of days designing a way that works for me. We can discuss further but here is what I found so far. @petergrossmann21 You may find this interesting as well. For context, I am also working with fixed devices with fixed IO; but different circuits have different fixed IO. I am working on improving the VTR task infrastructure to allow me to do this all within one config file (which it cannot do at the moment).

I agree with what @saaramahmoudi said above; an architecture can specify both an `auto_layout`  and `fixed_layouts`. The example I am working with locally:
[image]

As far as I can tell (I am still doing a bit more testing) when the `--device` argument is not specified, it will default to use the `auto_layout` in this case. From previous experience, if your architecture does not have an `auto_layout` specified, it will pick from the `fixed_layouts` if a device is not specified; but I have never checked this directly.

For what I am working on, I need to constrain different circuits to different devices and atom constraints in the same VTR task config file. I am working on introducing a new option to the config file called `circuit_constraint_list_add` which will "constrain" a circuit to a specific architecture, device, or constraint (it can be one or all of these). So it would look something like this:
[image]
Notice that in this example, I constrained the first three circuits to the `mcnc_min_device` and the last circuit to the `mcnc_medium_device`.

This is still WIP, but I hope to merge this in sometime in the next week or so.

petergrossmann21, 2024-10-17:
> From previous experience, if your architecture does not have an `auto_layout` specified, it will pick from the `fixed_layouts` if a device is not specified; but I have never checked this directly.

@AlexandreSinger  I can confirm that it will pick a fixed layout with no --device option specified and no auto_layout provided.

If anyone has interest in the basic approach to layout construction I am using, a pair of examples are available in the open 
[here](https://github.com/zeroasiccorp/logik/releases/tag/fpga-v0.1.22).  These are starting to show their age a bit in certain respects but for this discussion are still on point.

I do not currently take advantage of the ability to specify multiple layouts in the same arch file, so many of the considerations here are intellectual curiosities for my use case rather than practical considerations.  That said, I can imagine some very forward looking scenarios where diving deeper on this topic would be relevant for my work.

petergrossmann21, 2024-10-17:
Based on industry sync meeting discussion, I'm opting to close this issue; we can reopen a related subtopic as a future issue if it needs additional discussion.

closed 2024-10-17 by petergrossmann21