verilog-to-routing/vtr-verilog-to-routing #2811: Wire lookahead runtime scales poorly with number of switch/segment types

petergrossmann21, 2024-11-15:
During testing of the delay modeling approach I am developing (related to https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/2664, although I am not using that feature just yet to help stage the debugging process), my testing flow has matured to the point where I can comment more definitively on the impact of adding a large number of switch and segment types on router lookahead runtime.  This was flagged as a possible concern by @rs-dhow in previous discussions.

It appears that when the router lookahead algorithm has a large number of switch/segment types to manage (I am generating segments and switches in matched pairs, so the number of segment types and switch types in my architecture models are equal), the lookahead time does indeed increase.  I'm putting together some slides and additional experiments to describe the results in more detail.

#### Expected Behaviour

For a small architecture (1500 LUTs), I would expect little to no increase in router lookahead runtime if I increase the segment count by a significant amount.

#### Current Behaviour

For a small architecture (~1500 LUTs), I observe that when the switch list and segment lists counts increase from 2 to ~200, the router lookahead runtime increases from ~6 seconds to ~50 seconds and dominates total router runtime.  I expect to be able to show further degradation with larger architectures; experiments to quantify such a trend are in flight.

#### Possible Solution

I have not looked into possible solutions yet, but this is central enough to my IC development needs that I could invest time into developing a PR to resolve the issue.  A collaboration with more seasoned VPR developers is likely the most effective approach, even if I do most of the heavy lifting.

#### Steps to Reproduce

1. Please contact me to make arrangments for test case sharing, as I have not yet open-sourced all components used in generating the results described above.

#### Context

The goal of the delay model I am attempting to generate is to harvest ASIC EDA software output for use in delay annotation of the components of an eFPGA that will be productized.  The delay information is unique on a per tile-type, per routing channel basis, so the number of unique delay parameters is O(1K) to O(10K).  For accurate delay modeling, VPR's algorithms thus need to scale well as the number of segment types and switch types become large.

#### Your Environment

* VTR revision used:  de31f094aa4f894a5e6e0dc32c66365f4b341190
* Operating System and version:  Ubuntu 20.04
* Compiler version:  GCC 10.5.0

soheilshahrouz, 2024-11-17:
The size of map router lookahead grows lineaerly with the number of segment types.

https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/66f35d9f26255de8e01d68f0395b84ed006ca28f/vpr/src/route/router_lookahead_map.h#L43-L53

Router lookahead computation involves iterating over all segment types, sampling nodes for each type, and building cost maps. This likely explains why computation time increases as the number of segment types grows.

https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/66f35d9f26255de8e01d68f0395b84ed006ca28f/vpr/src/route/router_lookahead_map.cpp#L522-L554

Two potential solutions are:
- Compute cost maps for different segment types in parallel.
- Use a smaller set of representative segment types to generate the lookahead.

With the first solution, the memory usage of the router lookahead still increases linearly with the number of segment types. However, if each segment type is confined to a limited area of the device, this approach reduces reliance on the translation-invariance assumption in the router lookahead.

petergrossmann21, 2024-11-19:
@vaughnbetz a couple of follow-on notes:

1.  Upon further review of my setup, it turns out that the reported runtimes are for a build of vpr with BUILD_TYPE=debug.  I will need to do some reporting on results with a normal build as well; early indicators are that a standard build mutes this issue, although to @soheilshahrouz 's point the linear scaling will always be there.
2. That said, I think that from a hardware perspective allowing the RR graph to store different delays per wire rather than per wire type is much more intuitive.  The solution I have been prototyping feels like a workaround to cope with your proposed feature not being implemented.
3. I personally am not using R and C because my EDA flow for delay parameter collection lumps gate delays and wire delays together, so if I set R and C to a nonzero value I end up counting wire delay twice.  I can imagine that this may not be generally true.  A user wishing to adopt a per-wire delay annotation method where the intrinsic gate delay and wire delay are captured separately would want to be able to set R and C values on individual wires (for example, by parsing a SPEF file and then transferring its RC data to VPR).

tangxifan, 2024-11-22:
Upvote for the feature where each MUX has individual RC values. Prefer to have Tdel which offers an easy way to annotate. Using RC modeling will cause extra effort and increase difficulty level when performing timing extraction from layouts.

vaughnb-cerebras, 2024-11-28:
This should be 8x faster in recent builds (we changed the router lookahead builder config. But the discussion of how to handle many different wire delays (due to post-layout-extracted delays) is still a good one that we'll keep talking about in the weekly developer meeting.