verilog-to-routing/vtr-verilog-to-routing #2821: System Verilog support is broken due to compilation error in F4PGA plugin

MohamedElgammal, 2024-11-21:
The System verilog is broken as it requires enabling `YOSYS_F4PGA_PLUGINS` and doing so is getting a compilation error as follows:

vaughnbetz, 2024-11-25:
@soheilshahrouz : in case you feel like a tour of some yosys + C++ stuff ...

hzeller, 2024-11-25:
Isn't the systemverilog plugin called [synlig](https://github.com/chipsalliance/synlig) for a while now ? I suspect this pulls in a somewhat old version...

amirarjmand93, 2024-11-25:
I have been testing Synlig for integration, and so far it appears to be a suitable tool for SystemVerilog, rather than using it as just a standalone parser and elaborator, like Surlog. I will provide updates here along with some screenshots of my progress asap.

amirarjmand93, 2024-11-26:
[image]

@vaughnbetz I executed some SystemVerilog designs of the VTR flow benchmark. It passed from Synlig, Parmys, Yosys, and VPR end-to-end successfully. It seems OK, but it takes more time to integrate fully. I am trying to solve it.

vaughnbetz, 2024-11-26:
That's good news! Thanks @amirarjmand93.

amirarjmand93, 2024-12-04:
I am writing to provide an update regarding the integration of **Synlig**.

#1
 It seems that a race condition is causing a nondeterministic situation, leading to errors with some nested submodules during the synchronization and update process. In some cases, I encounter no errors with the same configuration.

#2
Sometimes, I encounter the error (RPC failed) and similar issues when cloning(non-fatal error), even though the network is functioning well and the Postbuffer size has been increased. 

[image]

It's in progress.

vaughnbetz, 2024-12-04:
The clone error sounds like a timeout/disconnect issue with the chipsalliance repo.  Adding @hzeller in case he knows why that repo is disconnecting before the clone is done. @amirarjmand93 : I imagine we could switch from a submodule to a subtree for this dependency if necessary, as I believe a subtree would actually put the files in the VTR repo, and hence cloning should happen from our repo and hopefully be robust.