verilog-to-routing/vtr-verilog-to-routing #2842: Incorrect constant K-LUT primitive instantiation

Junius00, 2024-12-11:
#### Expected Behaviour
To implement global GND and VCC signals, I understand that VTR makes use of K-LUTs to generate these constant signals. In theory, the LUT mask should be all '0's or all '1's, which outputs a '0' or '1' regardless of LUT inputs.

#### Current Behaviour
When using `--gen_post_synthesis_netlist`, the constant LUT for a global GND is instantiated as follows:
```
LUT_K #(
        .K(5),
        .LUT_MASK(32'b11111111111111111111111111111110)
    ) \lut_gnd  (
        .in({
            1'bX,
            1'bX,
            1'bX,
            1'bX,
            1'bX
         }),
        .out(\lut_gnd_output_0_0 )
    );
```
Which yields 2 problems:
1. The LUT mask is not as expected (all '0'), but rather **only the last output bit is set to '0'**.
2. The `LUT_K` primitive present in `vtr_flow/primitives.v` uses the input `in` as an array index of the `LUT_MASK` parameter. Setting them as `1'bX` generates an output signal of `1'bX` as a result, which messes with all modules relying on this global GND output.

#### Possible Solution
The only possible solution I've managed to get proper Verilog simulations out of is to **change the inputs to all '0's** and then ensure that the **last bit in the LUT mask is equal to the desired constant output**:
- Any undefined signal in `in` will cause an undefined `out`.
- If the `in` signal is fixed at a certain position, then only that bit in the LUT mask will matter.

#### Steps to Reproduce

1. Run VTR with `--gen_post_synthesis_netlist` with any circuit that requires global GND/VCC (e.g., simple addition with a constant).
2. Inspect the `*_post_synthesis.v` file.

#### Context

When trying to verify that the final implementation of VTR matches the initial Verilog design provided with a testbench, the undefined output of this constant K-LUT has a tendency to cause all the post-implementation module's outputs to be undefined, which leads to incorrect simulation results.

#### Your Environment

* VTR revision used: latest commit `ce706d579de8d370994fb3d3eb1a469d98707866`
* Operating System and version: Ubuntu 22.04.4 LTS
* Compiler version: cmake version 3.22.1, GNU Make 4.3

vaughnbetz, 2024-12-11:
Thanks @Junius00 !  @KennethKent @amirarjmand93 @tangxifan @duck2 : this looks high priority to fix, and from Junius' description it should be a matter of changing the output for gnd/vcc LUTs.  Any volunteer to fix this?  I'm surprised we haven't hit this before.

vaughnbetz, 2025-01-16:
Just pinging people on this ... @Junius00 : do you want to make a PR to fix this?  Our tests should catch if the fix has a bug, so if CI and the nightly tests pass it should be safe to merge.