verilog-to-routing/vtr-verilog-to-routing #2864: Documentation updates to remove a deprecated option (place_cost_exp) and add an undocumented one (place_constraint_expand)

vaughnbetz, 2025-01-16:
place_cost_exp has been removed, but it is still in the VPR command line options -- it should be removed.

place_constraint_expand isn't documented, but it should be (unless it's a very specific option intended only for developers, which is possible).

soheilshahrouz, 2025-01-17:
`place_constraint_expand` is used to generate placement constraints based on the output of the placement stage. I feel that this option was added to generate rational placement constraints for testing the floorplanning feature in VPR.

`place_cost_exp` is removed in #2863

vaughnbetz, 2025-01-20:
Thanks. The changes look good. Just to confirm: the dusty_sa annealing schedule has already been removed from the annealing code, correct?  (I did a quick search and didn't find it, so I think it's gone, in which case removing from the documentation in this PR makes sense).

vaughnbetz, 2025-01-22:
Closing as completed. And the dusty_sa annealing schedule is confirmed removed.

closed 2025-01-22 by vaughnbetz