verilog-to-routing/vtr-verilog-to-routing #2868: [LibVTRUtil] Confusing API for vtr::rect

AlexandreSinger, 2025-01-21:
While working on the initial placer in VTR, I found a bug when the code is performing centroid placement and is determining if a centroid placement is legal or not:
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/848d1e77d8ea222e6f84716172c12b0eb84d8347/vpr/src/place/initial_placement.cpp#L302-L333

The issue is that the code creates a partition region that is the size of the device for unconstrained blocks. For example, for an 11x11 architecture it creates a rectangle from (0, 0) to (10, 10). The problem is that this code uses the `contains` method from vtr::rect. This does not include the top-right edges; so any centroids which were placed on the top-right edge of the device were being wrongfully rejected. 

https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/848d1e77d8ea222e6f84716172c12b0eb84d8347/libs/libvtrutil/src/vtr_geometry.h#L171-L181

An easy fix for this problem is to use the `coincident` method instead; however, there may be other misunderstandings throughout VTR which need to be investigated.

In reality, this interface is a little confusing, and these methods should be renamed or even removed if they are not being used.

We will need to look through the code for cases where these methods are used and ensure they are correct.

AlexandreSinger, 2025-01-21:
@soheilshahrouz @vaughnbetz FYI, this was the issue that I found in the initial placer.

vaughnbetz, 2025-01-22:
Good catch, and this is bizarre.  I'm not sure why we'd want two definitions of contains/coincident.  I agree it should be at least renamed, and if there is no good use case for the current coincident vs. contains we should remove one (I vote for coincident) and have the other do the usual check where you include the edges.

vaughnbetz, 2025-01-22:
It might be that since these methods were defined with a template parameter someone was worried about the <float> version not checking for inclusion / equality (although I'm not sure why).  The behaviour doesn't make sense for an <int> though, and I suspect it doesn't make much difference for <float>