verilog-to-routing/vtr-verilog-to-routing #2879: Unable to generate odin_ii

aryan-1307, 2025-02-03:
Hi 
I was trying to run this code , but after compilation it was unable to generate OdinII because of which I was unable to map a Verilog file on any of the FPGA architecture. It is unable to create a blif file. I have checked for all error messages but I am unable to get any solution for this issue. I am trying to run it on wsl on latest version of ubuntu.
Also I tried making blip file manually but still there is another error showing that it has failed to load vpr even though I have written it’s correct path. 
So is it the problem of my system or anything else Is going wrong from my side?

AlexandreSinger, 2025-02-03:
Hi Aryan, can you show the commands you used and the errors produced?

Also, FYI, we recommend using the Parmys frontend instead of Odin-II. See the quick start guide for more information on the expected way to build VTR: https://docs.verilogtorouting.org/en/latest/quickstart/

aryan-1307, 2025-02-06:
Thanks! I followed the documentation provided.@AlexandreSinger 
I am still having trouble generating the Parmys executable after compiling several times. I have modified the CMakeLists.txt file for Parmys changing add_library to add_executable and removing the shared option since I saw a Parmys.so file after compilation I have also enabled the Parmys front end in the build configuration however, the parmys executable isn’t been generated and I can’t find it anywhere in the repository. I am not getting any specific error messages just this overall issue. What should I do next?

AlexandreSinger, 2025-02-07:
Did you follow the environment setup carefully? https://docs.verilogtorouting.org/en/latest/quickstart/#environment-setup

Also what system are you on? We currently support recent versions of Ubuntu.

Make sure you also remember to make: https://docs.verilogtorouting.org/en/latest/quickstart/#build-vtr

aryan-1307, 2025-02-10:
Here’s a revised version of your GitHub issue/chat message based on your requirements:

Chat with Contributor - Issue with parmys Executable and VPR in VTR Flow

Hi, I am setting up VTR on Ubuntu 24.04 (WSL) and facing issues with the parmys executable not being generated. I have followed the documentation exactly and executed the following commands so far:

# Cloned the repository
git clone --recursive https://github.com/verilog-to-routing/vtr-verilog-to-routing.git

# Initialized and updated submodules
git submodule init
git submodule update

# Installed dependencies
./install_apt_packages.sh

# Created and activated virtual environment
make env
source .venv/bin/activate
pip install -r requirements.txt

# Created a separate build directory and compiled VTR
mkdir -p build
cd build
cmake ..
make

# Checked the executables
ls ~/projects/verilog-to-routing/build/vpr/vpr   # Present
ls ~/projects/verilog-to-routing/build/abc/abc   # Present
ls ~/projects/verilog-to-routing/build/parmys    # Only parmys.so, no parmys binary

# Set correct paths
export PATH=~/projects/verilog-to-routing/build/vpr:~/projects/verilog-to-routing/build/abc:$PATH
export LD_LIBRARY_PATH=~/projects/verilog-to-routing/build/parmys:$LD_LIBRARY_PATH

# Checked if VPR and ABC run
vpr --version   # Works
abc -h          # No output

# Attempted to run VTR flow
source .venv/bin/activate
python vtr_flow/scripts/run_vtr_flow.py my_design.v my_arch.xml --route_chan_width 100

Errors Encountered:
	•	The parmys executable is missing, but parmys.so is generated.
	•	vpr --version works, but when running the VTR flow, it fails to load properly.
	•	Even after adding correct paths and activating the virtual environment, the issue persists.

I have also attached screenshots of the build directory and error messages for reference.
Am I going wrong anywhere? Any suggestions on fixing this?

Let me know if you need any changes before sending it! 🚀

AlexandreSinger, 2025-02-10:
Hi @aryan-1307 , thank you for the detailed message. A few issues with your approach:

1. You are building directly using CMake. This _should_ be fine; but many of the developers and users of VPR just use the MakeFile to build VTR instead of invoking CMake directly. I would recommend doing so as well just in case. This should be as simple as just typing `make` in the root directory of VTR. See https://docs.verilogtorouting.org/en/latest/quickstart/#build-vtr
2. The executables of CMake projects may not be in their respective build folder. If you are looking for generated executables, I recommend looking into the `build/bin` folder instead!
3. Parmys is not an executable (as far as I am aware), it is a flow built off of `yosys` (which is an executable). It can be very challenging to run the Yosys flow manually, so we created a script to do it for you! I recommend using this script if your goal is to use Parmys to synthesize a Verilog file into a blif file! https://docs.verilogtorouting.org/en/latest/quickstart/#synthesizing-with-parmys
4. You should not have to set you PATH variables for VTR. Our scripts know where the executables are.
5. The error you are showing for VTR flow is not a Parmys issue (as far as I can tell), it looks like an error occured in VPR instead: [image]
See the log file in this picture for a detailed error message from VPR with why it failed.

Hope this helps!