verilog-to-routing/vtr-verilog-to-routing #2883: Error while Building CXX object vpr/CMakeFiles/libvpr.dir/src/server/gateio.cpp.o

PranavKumar1408, 2025-02-06:
Hi
I get the following error while trying to 'make' the vtr9.0.0 release:
```~/vtr-verilog-to-routing-9.0.0/vpr/src/server/gateio.cpp: In member function ‘server::GateIO::ActivityStatus server::GateIO::check_client_connection(sockpp::tcp6_acceptor&, std::optional<sockpp::stream_socket_tmpl<sockpp::inet6_address> >&)’:
~/vtr-verilog-to-routing-9.0.0/vpr/src/server/gateio.cpp:59:51: error: conversion from ‘sockpp::result<sockpp::stream_socket_tmpl<sockpp::inet6_address> >’ to non-scalar type ‘sockpp::tcp6_socket’ {aka ‘sockpp::stream_socket_tmpl<sockpp::inet6_address>’} requested
   59 |     sockpp::tcp6_socket client = tcp_server.accept(&peer);
      |                                  ~~~~~~~~~~~~~~~~~^~~~~~~
~/vtr-verilog-to-routing-9.0.0/vpr/src/server/gateio.cpp: In member function ‘server::GateIO::ActivityStatus server::GateIO::handle_sending_data(sockpp::tcp6_socket&)’:
~/vtr-verilog-to-routing-9.0.0/vpr/src/server/gateio.cpp:79:52: error: cannot convert ‘sockpp::result<long unsigned int>’ to ‘std::size_t’ {aka ‘long unsigned int’} in initialization
   79 |             std::size_t bytes_sent = client.write_n(task->response_buffer().data(), bytes_to_send);
      |                                      ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                    |
      |                                                    sockpp::result<long unsigned int>
~/vtr-verilog-to-routing-9.0.0/vpr/src/server/gateio.cpp: In member function ‘server::GateIO::ActivityStatus server::GateIO::handle_receiving_data(sockpp::tcp6_socket&, comm::TelegramBuffer&, std::string&)’:
~/vtr-verilog-to-routing-9.0.0/vpr/src/server/gateio.cpp:116:48: error: cannot convert ‘sockpp::result<long unsigned int>’ to ‘std::size_t’ {aka ‘long unsigned int’} in assignment
  116 |         bytes_actually_received = client.read_n(&received_message[0], CHUNK_MAX_BYTES_NUM);
      |                                   ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                |
      |                                                sockpp::result<long unsigned int>
~/vtr-verilog-to-routing-9.0.0/vpr/src/server/gateio.cpp: In member function ‘server::GateIO::ActivityStatus server::GateIO::handle_client_alive_tracker(sockpp::tcp6_socket&, std::unique_ptr<server::GateIO::ClientAliveTracker>&)’:
~/vtr-verilog-to-routing-9.0.0/vpr/src/server/gateio.cpp:174:54: error: cannot convert ‘sockpp::result<long unsigned int>’ to ‘std::size_t’ {aka ‘long unsigned int’} in initialization
  174 |                 std::size_t bytes_sent = client.write(message);
      |                                          ~~~~~~~~~~~~^~~~~~~~~
      |                                                      |
      |                                                      sockpp::result<long unsigned int>
```

#### Expected Behaviour

Should make without any errors

#### Steps to Reproduce

1. Download the vtr9.0.0 official release and extract it
2. cd into vtr directory
3. > ./install_apt_packages.sh
4. > pip install -r requirements.txt
5. > make

#### Context
Attempting to build vtr9.0.0

#### Your Environment
* VTR revision used:VTR9.0.0
* Operating System and version:Ubuntu 22.04.5 LTS x86_64
* Compiler version:gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0

AlexandreSinger, 2025-02-07:
@PranavKumar1408 Thank you for flagging this issue! It looks like the zip files for VTR 9 do not include the external libraries we use in VTR!

To quickly get around this issue, you can do the following steps:

- Clone the VTR9 branch: https://github.com/verilog-to-routing/vtr-verilog-to-routing/tree/v9.0.0
- Follow the QuickStart guide to set up your enviornment and build VTR: https://docs.verilogtorouting.org/en/latest/quickstart/#

Importantly, make sure you do `git submodule init` and `git submodule update` after you clone to get the correct version of the libraries that VTR needs.

PranavKumar1408, 2025-02-08:
Hey @AlexandreSinger,
Cloning the VTR9 branch worked perfectly.
Thanks!

vaughnbetz, 2025-02-08:
Thanks for flagging this!  Re-opening and assigning to @MohamedElgammal so we can update the zip file to also include these files, making it more convenient for the future.

vaughnbetz, 2025-02-08:
@MohamedElgammal : at the same time you regenerate the v9 zip file, it would be good to pull in the two recent commits to v9 that fixed some issues.

vaughnbetz, 2025-03-31:
@MohamedElgammal : nag, nag ...

AlexandreSinger, 2025-06-26:
Has this issue been resolved yet? I received an email from another researcher who also had issues with building the VTR9 release.

AlexandreSinger, 2025-06-26:
Thanks @MohamedElgammal , I think you are correct. There is a clear warning to use the proper zip files in the release page. I have verified that the new files build successfully without issue.

MohamedElgammal, 2025-06-26:
Thanks for confirming it. I will close this issue then.