verilog-to-routing/vtr-verilog-to-routing #2907: [CI] Update Golden Results for Nightly Tests 1 and 3

AlexandreSinger, 2025-02-24:
Nightly Tests 1 and 3 are currently failing QoR on a few tests after a recent PR (PR #2897 ):

[image]

[image]

See: https://github.com/verilog-to-routing/vtr-verilog-to-routing/actions/runs/13486749110

Only the golden results for these two tests (`vtr_reg_nightly_test1/arithmetic_tasks/multless_consts` and `regression_tests/vtr_reg_nightly_test3/vtr_reg_qor_chain_predictor_off`) need to be regenerated. It looks like the min channel width increased slightly which should not be an issue.

AlexandreSinger, 2025-02-24:
@haydar-c Can you update these golden results?

haydar-c, 2025-02-24:
Sure!

closed 2025-02-28 by AlexandreSinger