verilog-to-routing/vtr-verilog-to-routing #2940: 'make format' looks specifically for clang-format version 14

AmirhosseinPoolad, 2025-03-19:
When running 'make format' while having any version of clang-format other than 14, make format does not work.

#### Expected Behaviour
Any clang-format installed by user should work

#### Current Behaviour
If anything other than version 14 is installed, 'make format' would not work

#### Error Log
```
Building target(s): format
xargs: clang-format-14: No such file or directory
make[4]: *** [CMakeFiles/format-cpp.dir/build.make:70: CMakeFiles/format-cpp] Error 127
make[3]: *** [CMakeFiles/Makefile2:918: CMakeFiles/format-cpp.dir/all] Error 2
make[2]: *** [CMakeFiles/Makefile2:1004: CMakeFiles/format.dir/rule] Error 2
make[1]: *** [Makefile:283: format] Error 2
make: *** [Makefile:94: format] Error 2
```

AlexandreSinger, 2025-03-22:
@AmirhosseinPoolad Was this resolved in your recent PR?

AmirhosseinPoolad, 2025-03-22:
Oh, yes it was fixed, thanks for mentioning!

Fixed by #2945

closed 2025-03-22 by AmirhosseinPoolad