verilog-to-routing/vtr-verilog-to-routing #3015: GUI not showing wires between connected CLBs despite highlighting

andrecavalcante, 2025-05-01:
Hi, apologies if this isn't an actual issue. In post-routing netlist files, the signals between clusters are visible. However, when using the GUI's "Toggle nets" feature to visualize these connections, some signals do not appear under cluster nets—they only show up under primitive nets. In these cases, clicking on a CLB highlights the connected CLBs, but no wire (cluster net) is displayed between them. So are there real wires between blocks connected through primitive nets? If not, is it possible to infer this from the post-routing netlist file?

Thanks!

AmirhosseinPoolad, 2025-05-08:
Thank you for filing this issue! Can you provide the steps to reproduce this issue? Specifically the command you used to run VPR and the steps you take in the GUI to see this behaviour happen.

vaughnbetz, 2025-05-12:
From a quick look, it looks like we draw the nets between clusters using the real routing resources (after routing), but draw the connections between primitives within the clusters as flylines only, and draw the connections from primitive-to-primitive that go across clusters as flylines too.  The last part is a bit confusing.
I think we should use this issue as a driving issue to show flat routing, and combinations of between-cluster and within-cluster connectivity, nicely.  @SamuelHo10 : first step is playing with the current graphics, and the second is making a plan we can review.