verilog-to-routing/vtr-verilog-to-routing #3189: Exploration of Using a Different Memory Allocator for VTR Runtime Improvement

stefanpie, 2025-07-05:
Note, this is an independent exploration that I conducted, and I don't expect that VTR authors to immediately consider these changes in VTR. I merely want to provide some useful insight into VTR I gained from my own research to the authors of VTR to help make VTR better.
## Motivation

Recently, I've been looking at ways to speed up EDA tools without the need to make changes to the underlying code of the EDA tools themselves. For example, I want to try to make Vitis HLS or Vivado run faster, but I don't have access to the Vitis HLS or Vivado source code, and I can't change it unless I go work or intern for AMD/Xilinx. I want to explore approaches that would let me speed up commercial EDA tools in a "black box" manner. My original intent with this idea was that I could speed up EDA tools that interact with LLM agents, since the bottleneck there is the EDA tools (which can take >30s to minutes) and not LLM inference (which is usually <20s). But this is also a useful pursuit in general for large-scale EDA dataset collection and EDA benchmarking.

## Approach

One trick in this exploration was to use a different memory allocator than the default malloc implementation that was compiled or linked with the software. For my exploration, I was testing the use of mimalloc (https://github.com/microsoft/mimalloc), a custom memory allocator from Microsoft Research, which has many tunable and customizable parameters. Even better, you can override the default allocator used by any software that dynamically links malloc using LD_PRELOAD to swap in mimalloc instead.

The general idea here is that (assuming our system has lots of memory resources) we can tune the memory allocator to take advantage of memory resources to trade off faster program execution for a small amount of memory overhead. This can hopefully result in non-trivial speedups in runtime over many program executions. For example, if I have to place and route 1000 benchmarks and I can get a 0.9x-0.8x reduction in runtime with a better memory allocator, my benchmark evaluation can take 9-8 hours instead of 10 hours. However, this model of what I would expect speedups to be is not that realistic. Program execution and memory allocation in a tool may be complex. Running the program on small designs might yield a 0.99x reduction (virtually nothing) while running the program on larger designs might yield a 0.8x reduction. Some programs or phases of program execution (synth vs. pack vs. place vs. route) might also see different behaviors and runtime reductions as well.

I decided to also try to evaluate this approach on open-source EDA software like yosys (still testing) and VTR, and I was able to capture some simple initial results for VTR just by swapping in mimalloc with minimal tweaking of the allocator configuration.

## Evaluation

I benchmarked the VTR execution using two designs:
- `mcnc_simple`: `$VPR_BIN --pack --place --route --analysis --write_rr_graph rr_graph.xml -j1 ./EArch.xml ./sqrt8.blif` 
- `mcnc_big`: `$VPR_BIN --pack --place --route --analysis --route_chan_width 128 --write_rr_graph rr_graph.xml -j1 ./EArch.xml ./clma.blif` 
- `mcnc_big_search`: `$VPR_BIN --pack --place --route --analysis --write_rr_graph rr_graph.xml -j1 ./EArch.xml ./clma.blif`

Each case was executed with 48 samples each, all running in parallel on 48 cores (on a 64 core machine), with each VTR run using no parallelism inside VTR. During the execution there, the total used memory was always way lower than the total system memory, so there should have been no weird memory throttling or swapping behavior.

The figures below show the results. For each execution, there is a base version where the design files are read and written to disk, a version where design files are read and written to memory filesystem (using the shm mount), and the case where files are accessed from a memory file system and also mimalloc is dropped in place with minimal tweaking of the configuration. The runtimes of each of these cases is plotted as a "histogram" (really a KDE plot).

[image]

[image]

[image]

I also included some tables showing the normalized runtime for each method for the mean, median, 5%-th percentile and 95%-th percentile of runtimes. These are all normalized to the "base" approach which means files on disk and no mimalloc used.

**Results for `vtr__mcnc_simple`**

| Method       |   Mean |   Median |   5% Quantile |   95% Quantile |
|:-------------|-------:|---------:|--------------:|---------------:|
| disk         |   1    |     1    |          1    |           1    |
| shm          |   0.85 |     0.91 |          0.64 |           0.89 |
| shm+mimalloc |   0.92 |     0.93 |          0.78 |           0.95 |

**Results for `vtr__mcnc_big`**

| Method       | Mean | Median | 5% Quantile | 95% Quantile |
| :----------- | ---: | -----: | ----------: | -----------: |
| disk         |    1 |      1 |           1 |            1 |
| shm          | 0.97 |   0.92 |        1.02 |         0.96 |
| shm+mimalloc | 0.98 |   0.94 |        1.03 |         0.97 |

**Results for `vtr__mcnc_big_search`**

| Method       | Mean | Median | 5% Quantile | 95% Quantile |
| :----------- | ---: | -----: | ----------: | -----------: |
| disk         |    1 |      1 |           1 |            1 |
| shm          | 0.97 |   0.92 |        0.92 |         1.06 |
| shm+mimalloc | 0.96 |   0.89 |        0.94 |         1.04 |

I also conducted some (non-rigorous) non-parametric statistical tests (since I don't know if I can assume normality) to see if there is a statistically significant difference in runtime. Although I would not draw any conclusive results from these results without many more samples, it's just a nice thing to set up in the code that is useful once I can test many more samples across more designs.
 
```
# Statistical Analysis Report: vtr__mcnc_simple
## Mann-Whitney U Test Results
- SHM < Disk: U-statistic = 89.00, p-value = 0.0000
- SHM + Mimalloc < Disk: U-statistic = 197.00, p-value = 0.0000

# Statistical Analysis Report: vtr__mcnc_big
## Mann-Whitney U Test Results
- SHM < Disk: U-statistic = 467.00, p-value = 0.2751
- SHM + Mimalloc < Disk: U-statistic = 485.00, p-value = 0.3610

# Statistical Analysis Report: vtr__mcnc_big_search
## Mann-Whitney U Test Results
- SHM < Disk: U-statistic = 447.00, p-value = 0.1932
- SHM + Mimalloc < Disk: U-statistic = 405.00, p-value = 0.0764
```
## Conclusion

Looking at the simple results so far, it seems like there is a compelling case here to be made to explore further the use of an alternative memory allocator, especially mimalloc. The initial results on three execution VTR cases with different designs and minimal tweaking to mimalloc suggest that 1) with minimal changes we can get ~0.9x reduction in runtime 2) more benchmarking across many more designs and execution modes of VTR is needed and 3) more exploration of how to configure mimalloc for VTR is also needed. Since VTR is open-source, unlike the commercial EDA tools that motivated this exploration, I can actually also gain some insight from the VTR code itself to help guide the configuration of mimalloc. I also want to note that I plan to run further experiments to disentangle the effects of the other techniques I explore, using an in-memory file system, from the technique I propose here, using a different memory allocator. In the current results, there are no results for just adding mimalloc when files are still being read from and written to disk. But hopefully, as I explore other black-box EDA tool speedup techniques, I can update my VTR results alongside my other results from other tools.

Hopefully this provides some useful insights for the VTR authors, and I am definitely open to discussion on this idea!

stefanpie, 2025-07-05:
For reproducibility and reference, here is the repo with all my code I used to create these results. As I work on these EDA tool speedup techniques and more benchmarking, I'll update the repo with more results and code.

https://github.com/stefanpie/faster-hls

Note, I will also at some point likely update the repo name to be more generic than just HLS. I started by messing with HLS tools, so that's why it's called `faster-hls`.

stefanpie, 2025-07-06:
**Important update since I made a mistake in my original issue post, but a good mistake in favor of good results.**

The initial results I shared are janky since mimalloc was never enabled even for the "shm+mimalloc" cases. I was using my portable statically linked version of VTR that I built, and since that statically links malloc from GLIBC, the LD_PRELOAD does not do anything, and mimalloc is never swapped in. This means that any improvement seen in the original results I posted is just by coincidence (maybe CPU warmup or some other system in the OS somewhere making it slightly faster).

However, this is a good mistake to have as I expect (and show below) that actually enabling mimalloc will show much more significant differences and would only speed up the program rather than make it slower.

To fix this, I also added another build to my portable VTR builds (see issue #3122 and my repo https://github.com/stefanpie/portable-vtr) with mimalloc statically linked overriding malloc from GLIBC. I then collected more results, but this time, anytime there is a test case with mimalloc and VTR, I use the new portable `vpr-mimalloc` build. For the normal VTR test cases with no mimalloc, I just use the original portable `vpr` build with malloc from GLIBC.

I also added a test case for using mimalloc but reading and writing files from disk. This means there are now four test cases: ("disk", "in-memory") x ("default malloc", "mimalloc").

Below are the updated figures and metrics.

[image]

[image]

[image]

**Results for vtr__mcnc_simple**
| Method        |   Mean |   Median |   5% Quantile |   95% Quantile |
|:--------------|-------:|---------:|--------------:|---------------:|
| disk          |   1    |     1    |          1    |           1    |
| shm           |   0.82 |     0.85 |          0.9  |           0.72 |
| shm+mimalloc  |   0.95 |     1.01 |          0.98 |           0.8  |
| disk+mimalloc |   0.86 |     0.92 |          0.89 |           0.74 |

**Results for vtr__mcnc_big**
| Method        |   Mean |   Median |   5% Quantile |   95% Quantile |
|:--------------|-------:|---------:|--------------:|---------------:|
| disk          |   1    |     1    |          1    |           1    |
| shm           |   0.82 |     0.84 |          0.78 |           0.86 |
| shm+mimalloc  |   0.96 |     0.98 |          0.9  |           0.99 |
| disk+mimalloc |   0.84 |     0.84 |          0.78 |           0.86 |

**Results for vtr__mcnc_big_search**
| Method        |   Mean |   Median |   5% Quantile |   95% Quantile |
|:--------------|-------:|---------:|--------------:|---------------:|
| disk          |   1    |     1    |          1    |           1    |
| shm           |   0.62 |     0.62 |          0.62 |           0.62 |
| shm+mimalloc  |   0.99 |     1.01 |          0.92 |           1.01 |
| disk+mimalloc |   0.61 |     0.63 |          0.59 |           0.62 |

```
# Statistical Analysis Report: vtr__mcnc_simple
## Mann-Whitney U Test Results
- SHM < Disk: U-statistic = 1070.00, p-value = 0.2752
- SHM + Mimalloc < Disk: U-statistic = 540.00, p-value = 0.0000
- Disk + Mimalloc < Disk: U-statistic = 413.00, p-value = 0.0000

# Statistical Analysis Report: vtr__mcnc_big
## Mann-Whitney U Test Results
- SHM < Disk: U-statistic = 992.00, p-value = 0.1213
- SHM + Mimalloc < Disk: U-statistic = 336.00, p-value = 0.0000
- Disk + Mimalloc < Disk: U-statistic = 315.00, p-value = 0.0000

# Statistical Analysis Report: vtr__mcnc_big_search
## Mann-Whitney U Test Results
- SHM < Disk: U-statistic = 1179.00, p-value = 0.5799
- SHM + Mimalloc < Disk: U-statistic = 0.00, p-value = 0.0000
- Disk + Mimalloc < Disk: U-statistic = 0.00, p-value = 0.0000
```

From these results, we do see real and very significant runtime reduction: ~0.9x on small designs that take less than a second to run, 0.85x on large designs that take on the order of 30s to run, and ~0.6x on large designs on the order of minutes where repeated implementation is done to search for the minimum channel sizes. These corrected results suggest that this approach is worthwhile and does result in noticeable runtime reductions. Also note that there is still some tweaking with mimalloc itself that can be configured at runtime (even when statically linked, you can still set mimalloc config values via env vars), which are still to be explored.

Sorry for the confusion with my original results, but it's good to see that proper integration of mimalloc shows significantly better results than expected!

stefanpie, 2025-07-09:
I have one more update I wanted to share. After tweaking the data collection to only allow 16 concurrent processes on a 64-core machine to collect 128 samples per test case, the results are much clearer now. This is in comparison to before where I was collecting 128 samples per test case but using 48 concurrent processes on our 64-core machine. I believe the pressure on the CPU scheduler and also contention with some other processes other users were running caused much more interference in the previous results I reported.

As shown below, now the improvement is much clearer for all VTR test cases.

[image]
[image]
[image]

What's also cool is that now, for the tiny design that runs the implementation in under a second, we can see the effect of reading the Verilog and architecture XML files from different file systems much more clearly at this scale. The in-memory file system is faster and more consistent than reading from disk.

Also, for considering an "end-to-end VTR flow" that uses Yosys, I also compiled a portable static build Yosys and abc to also use mimalloc and compared the results of running a Yosys script with the `synth_xilinx -family xc7 -abc9 ...` command as the main synthesis and tech. mapping step (this will technically call abc from Yosys as well, which is why I had to rebuild both with versions that use mimalloc).

We also see significant speed-ups (~0.8x) with Yosys/abc as shown below.

[image]
[image]

This Yosys/abc build and results thereof will solve the issue of making sure end-to-end tool chain speed-ups in VTR using this technique don't suffer from Amdahl's law. Just speeding up `vpr` without also speeding up `yosys` and `abc` is not as productive as speeding up both. I am ignoring Odin II for now, but getting that built the same way with mimalloc and being able to load it into the portable mimalloc version of Yosys is just more build scripting tweaks that I don't foresee being an issue.

stefanpie, 2025-08-02:
Some new updates since I got some time to work on this project a bit more.

I got the Titan benchmarks downloaded and running to start collecting numbers on implementing larger designs targeting larger FPGAs using VTR. I'm using the titan23 set of designs (described in the original Titan paper) for my experimentation, targeting the Stratix 10 (the architecture capture is described in the Titan 2.0 paper). For the Stratix 10 architecture, I am using the fixed layout corresponding to the 1SX040HH1F35E1VG part (146 x 148 tiles).

I ran implementation on the `neuron` design using the `.blif` and `.sdc` files shipped with the benchmark. The VPR command I ran mirrors what the Titan benchmarking Python script does when it runs.

```
./vpr \
    stratix10_arch.timing.xml \
    benchmarks/titan23/neuron/stratix10/netlists/neuron_stratix10_arch_timing.blif \
    --sdc_file benchmarks/titan23/neuron/stratix10/sdc/vpr.sdc \
    --max_router_iterations 400 \
    --router_lookahead map \
    --initial_pres_fac 1.0 \
    --router_profiler_astar_fac 1.5 \
    --seed 3 \
    --route_chan_width 400
```

For the case where I want to use the mimalloc version of VPR, I run the following:

```
MIMALLOC_ARENA_EAGER_COMMIT=1 MIMALLOC_PURGE_DELAY=2000 MIMALLOC_VERBOSE=1 ./vpr-mimalloc \
    stratix10_arch.timing.xml \
    benchmarks/titan23/neuron/stratix10/netlists/neuron_stratix10_arch_timing.blif \
    --sdc_file benchmarks/titan23/neuron/stratix10/sdc/vpr.sdc \
    --max_router_iterations 400 \
    --router_lookahead map \
    --initial_pres_fac 1.0 \
    --router_profiler_astar_fac 1.5 \
    --seed 3 \
    --route_chan_width 400
```

Note that I am using VTR 9.0, but with settings to run the Titan benchmarks that were decided when using an earlier version of VTR. This does not take advantage of cutting edge features like AP and flat placement, legalization, and so on.

Also note that I have not run this on our research server yet where the other results come from—just my Linux workstation—but I’m porting it over soon to gather more Titan data.

The initial results from both runs are shown below.

```
Normal VPR: The entire flow of VPR took 341.22 seconds (max_rss 3018.2 MiB)
Mimalloc VPR: The entire flow of VPR took 330.36 seconds (max_rss 2822.6 MiB)
```

I ran this again to confirm the consistency, and the second set of runs are basically the same numbers, +/- 2 seconds and +/- 1 MB.

What's interesting here is that the speedup is less significant—on the order of 0.96x reduction in runtime (or a 1.03x speedup if you compute it the inverse way). I also noted that for this design, the packing stage took up a huge chunk of time as noted in the original Titan paper.

However, what's more confusing is that in the VTR 9.0 paper the same neuron design takes a total of 583 seconds to run pack+place+route (from Table 17). But I can't trace down the configuration and exact settings used to create the Titan numbers in the VTR 9.0 papers. I might try to get more clarification of these details in one of the future public meetings so that I have a better idea of how to go about benchmarking these larger Titan designs and have some reasonable runtime numbers to compare to. Along the same lines, I’m going to try to look into benchmarking default flow settings to "presynth/pretechmapped" Titan blifs, but using newer versions of Yosys+Parmys as well in the VTR flow, as well as new VPR flow features not available when the original Titan or Titan 2.0 benchmarks were evaluated at the time of paper release.

Looking at the speedups I was getting before for VPR using mimalloc, they were ~0.84x for the large MCNC design for a single implementation pass, and 0.62x for the large MCNC design when allowing the implementation to search for minimal valid channel widths and minimal valid device sizes. These new numbers of 0.96x for the large fixed simple pass implementation of the Titan design imply two interesting things: 1) the scaling of design size and speedup is non-trivial (or at least non-constant) and requires more benchmarking, and 2) the speedup of using a different memory allocator might come more from placement and routing stages, or at least in implementations that require re-placement and re-routing. The second point is more interesting at the moment since the Titan implementation is packing-heavy, while the other implementations are balanced or skewed so most of their time is spent on placement and routing. Also, with iterated placement, every new place or route attempt might be freeing and reallocating large data structures associated with each place and route attempt, which is where the ~0.6x reduction might be coming from. In any case, more exploration is needed.

I also didn’t previously log max RSS for memory usage, but it seems like the new memory allocator makes it smaller, which is good. This is the first time I’m paying attention to memory overhead, so more on that to come.

So to summarize the update: it was a good suggestion to evaluate Titan, since there is new behavior showing up in the results as well as new interesting results for memory overhead. This is part of a larger evaluation I’m doing, so I have some Vivado numbers for a super simple design with the numbers showing virtually no speedup (~0.98x). Hopefully, I'll also get some Quartus numbers soon and maybe even Libero or ISE numbers if I feel like torturing myself trying to install those tools.

---

Quick aside: if you only want to get the blifs for the Stratix 10 from the Titan tar.gz, the command below works best for not unzipping the entire 150GB+ of data (you can sub in `gz` for `pigz` and don’t have to use `pv` if you don't want).

```
pigz -dc titan\_release\_2.0.0.tar.gz | pv | tar xf - --exclude='*.vqm' --exclude='*.log' --exclude='quartus2\_proj' --exclude='agilex' --exclude='stratixiv'
```

This leaves a 52GB set of data for the Stratix 10 instead, which is much nicer to work with and faster to extract.

gigeresk, 2025-08-04:
@Stefan, I ran some experiments comparing glibc malloc, tcmalloc, mimalloc, and jemalloc. I didn't run more than one design. I ran with 8 threads. The data is here: https://docs.google.com/spreadsheets/d/1t-GDES_IVs0lJqzKtnFL7Hg2Ym9pZHsxWgkHU1Mno5c/edit?usp=sharing
On this one datapoint, jemalloc is the best, followed by mimalloc. Runtime is better on mimalloc if you look at VPR's runtime collection.
I would encourage you to use jemalloc and multiple threads in your experiments.

stefanpie, 2025-08-05:
This is awesome! Thanks for taking the time to test it independently. Based on your data, it definitely seems worth it to test different allocators. I'll spend some time to rejig my benchmark code to parameterize across these allocators + my current effort to parameterize across Titan designs for the VTR subset of my experiments.

My main concern is that since I want to do all this on a "portable" staticly linked build of VTR, I am currently only supporting single threaded execution. I still need to follow up on being able to support statically comping with parallel support for routing + analytical placement + timing analysis. The limitation is something to do with Intel TBB library maybe not supporting static linking, but there was some discussion at the meeting that some parallel parts of VTR don't depend on TBB. I need to look into the VTR codebase deeper to untangle this a bit, but I think it's doable.

gigeresk, 2025-08-05:
Hope there's a clear winner. I'm motivated to get something merged on the default dynamically linked build; I'm happy to build off this work in due time.

closed 2026-08-21 by github-actions[bot]