verilog-to-routing/vtr-verilog-to-routing #3242: There might be three coding problems

victroix, 2025-08-21:
1.
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/246053dfea69af32e7ff0fdbe3c9b702b38a7bf1/vpr/src/place/delay_model/compute_delta_delays_utils.cpp#L194
    const size_t high_y = (longest_length <= device_height) ? std::max(device_width - longest_length, mid_y) : mid_y;
Should be: const size_t high_y = (longest_length <= device_height) ? std::max(device_height - longest_length, mid_y) : mid_y;
2.
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/246053dfea69af32e7ff0fdbe3c9b702b38a7bf1/libs/libvtrutil/src/vtr_math.h#L42
    auto iter = first + len / 2;
Might add after this line: if(len == 2)  {iter = first;}
at least i add these when generating delay table with reducer::median. It might be due to the fact that I'm using VS2015 for compilation.
 i tested if reducing two delays by median, for example, [0.8, 0.9], delay was 0.45, not 0.85.
3.
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/246053dfea69af32e7ff0fdbe3c9b702b38a7bf1/vpr/src/place/move_generators/simpleRL_move_generator.cpp#L234
        last_action_ = action_type_q_pos;
might be:         last_action_ = std::min((size_t)action_type_q_pos, num_available_actions - 1); ?

AlexandreSinger, 2025-08-21:
Hi @victroix thanks for the issue!

(1) certainly looks like a bug for me, @amin1377 what do you think?

(2) I do not think this issue was ever actually being hit. Due to some confusing code in vtr::math a different median function was being used. I have raised a PR ( #3243 ) which resolves this issue anyways since this method was definately broken.

(3) I am not sure what is the issue with this line, can you please elaborate what you think is wrong?

AlexandreSinger, 2025-08-22:
All issues have been resolved. Closing issue.

AlexandreSinger, 2025-08-22:
@victroix Thanks again for pointing these issues out!

closed 2025-08-22 by AlexandreSinger

victroix, 2025-08-22:
@AlexandreSinger @vaughnbetz Thank you for your reply. It is my honor to communicate with seniors from the VTR project.