verilog-to-routing/vtr-verilog-to-routing #3254: [CI] Strong tests not failing if there's a mismatch in architecture in golden results vs test results

AmirhosseinPoolad, 2025-08-25:
The architecture name for strong_scatter_gather golden results used to be different than the actual architecture file (Changed to arch file name without regenerating golden results) and this passed CI without while actually printing out an error.

The piece of code responsible for it is in parse_vtr_task.py:428. The issue (I think) is that the run_vtr_task file actually returns with a zero exit code:

[image]

(Blue circle next to the command means the return code of the command was zero)

How to reproduce:

change the architecture name in 'vtr_flow/tasks/regression_tests/vtr_reg_strong/strong_scatter_gather/config/golden_results.txt' to anything else, run './run_vtr_task.py regression_tests/vtr_reg_strong/strong_scatter_gather' then './run_vtr_task.py regression_tests/vtr_reg_strong/strong_scatter_gather -check_golden'

vaughnbetz, 2025-08-25:
Good catch ... can you track this down @AmirhosseinPoolad ?

vaughnbetz, 2025-08-25:
My guess is that the --check_golden option is not returning an error when it doesn't find the golden results file (even though it prints an error).  I believe it is returning a non-zero exit status when things don't match, or we shouldn't have gotten our various QoR test failures in the past.  So hopefully that's a hint of where to look.