verilog-to-routing/vtr-verilog-to-routing #3275: [Draw][Place] Manual Moves

AlexandreSinger, 2025-09-16:
I do not believe that manual moves are working as intended.

Currently, the `try_swap_` method takes a boolean argument which decides whether manual moves are on for the given swap or not. This boolean is only ever set to true during the initial temperature estimation as far as I can tell:
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/112d358b0c5a848720b6a514746d8adea157587b/vpr/src/place/annealer.cpp#L312-L321

This happens before any annealing iteration occurs. During the actual inner-loop of the annealer, you can see that this is always set to false:
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/112d358b0c5a848720b6a514746d8adea157587b/vpr/src/place/annealer.cpp#L683-L690

I wonder if something was removed from the inner-loop method which no longer allows for manual moves during the actual annealing algorithm.

See #3271 for context.

AlexandreSinger, 2025-09-16:
@vaughnbetz FYI

vaughnbetz, 2025-09-16:
Whoops, meant @haydar-c . Thank you for flagging the mistake!

haydar-c, 2025-09-26:
Sure, closing. Thanks @AlexandreSinger

closed 2025-09-26 by haydar-c